gdombiak/OctoPod #17: Add support for multi-languages

gdombiak, 2018-07-20:
OctoPod is being downloaded all around the world. We need to add the ability to translate the app to other languages.

Add Spanish support as a test.

ArtCC, 2018-09-22:
Hi!

I was going to translate the application into Spanish but I see that you do not use the Localizable.string file, all the texts are blocked in the storyboard...

Regards. Arturo.

gdombiak, 2018-09-22:
Hola Arturo,

You are correct. I'm now finishing last touches for the next 1.8 release and then start work on this issue. When I do this, I will need to add localization to soooo many places. Will let you know when ready to be translated.

Saludos
Gaston (de Argentina)

ArtCC, 2018-09-22:
Ok. Do you speak spanish? Are you from Argentina?

Saludos.

gdombiak, 2018-09-22:
Sip, de Buenos Aires. Me vine para usa hace unos 10 años. Vos de donde sos?

gdombiak, 2018-09-29:
Hola Arturo, recién hice un push con cambios para agregar lenguajes. Para probar las cosas agregué Español (de Latino America). Fijate como anda para España y si queres agrega la versión correcta para España. Avisame si ves algun error en mi traducción.

Gracias.
Gaston

gdombiak, 2018-09-29:
Genial. Gracias

gdombiak, 2018-09-29:
Strings have been localized. Creating new issues for each new language

closed 2018-09-29 by gdombiak