gdombiak/OctoPod #33: Camera Control

bdelia, 2018-07-25:
it would be nice to invert or rotate the camera via software to compensate for orientation like Octoprint allows for in the WebCam settings as well as the Octolapse plugin.

Dernjo, 2018-07-27:
I second this addition in the settings or as an alternative idea, for the app to read the settings in OctoPrint regarding camera orientation. But I love how you can tap on the camera feed and it enlarges to the full screen. Great job man!

gdombiak, 2018-07-27:
Yep, this is a cool idea. Was thinking of adding "Orientation" settings when entering information about the printer. Each printer might have their camera in a different orientation. Same for controlling Z orientation ( #34 )

gdombiak, 2018-07-28:
Implementation reads orientation from OctoPrint settings to decide how to render the camera. It also  reacts to live changes in Settings and updates the orientation accordingly.

closed 2018-07-28 by gdombiak

jeremyrclark, 2019-12-07:
How do you specify orientation on secondary cameras? It seems to respect my orientation from Octoprint's main camera setting, but I use MultiCam to display a secondary camera and I don't seem to have a way to flip/rotate that one.

gdombiak, 2019-12-16:
Hi @jeremyrclark,

The ideal situation would be for the MultCam plugin to let you configure/describe the orientation of each camera. OctoPod would then make use of that information to properly render each camera. What do you think of [creating an issue for the MultCam plugin](https://github.com/mikedmor/OctoPrint_MultiCam/issues) to see what the developer of the plugin thinks?

If that information is missing, do you have any ideas on possible workaround? 

Thanks,
Gaston

gdombiak, 2019-12-16:
It seems like [that request](https://github.com/mikedmor/OctoPrint_MultiCam/issues/2) was made back on May of 2018.

jeremyrclark, 2020-06-21:
The multicam plugin developer has added support for flipping the camera feed:
https://github.com/mikedmor/OctoPrint_MultiCam/issues/2

I'm able to flip the camera and it shows as expected in Octoprint, but in Octopod it remains upside down. Any chance you're able to leverage the new setting in Multicam?

gdombiak, 2020-06-22:
Hey @jeremyrclark, thanks for reminding me of this new cool improvement. I noticed it last week in the change log of the plugin and forgot to create a ticket for this. File this one: https://github.com/gdombiak/OctoPod/issues/348

Thanks,
Gaston

jeremyrclark, 2020-06-22:
Excellent @gdombiak - thanks for picking it up and looking forward to it!

commit 99a8fcf6 (2018-07-28): Use camera orientation that is configured in OctoPrint.

Closes #33