gdombiak/OctoPod #35: Interrupted print when connecting

Dernjo, 2018-07-27:
So I just installed you app and I was 70% through a print when I connected to octoprint via the app. After I connected the printer communication to ocotprint was interrupted causing the printer to go out of sync, stop receiving commands which was still being sent by octoprint and the failure of the print. 

I think and I’m probably wrong, when you connect to octoprint via the app it actually sends a comment to octoprint to connect to the printer, which causes it to try and connect again even if it is already connected. If this is true that it would interrupt the communication between the printer and octoprint, which would cause the printer to reset when a new connection is made and cause it to go out of sync. 

Or it could be something else that cause this problem but I figured I would post this as this is what happened when my print failed.

gdombiak, 2018-07-27:
Thanks @Dernjo for reporting this. Many controllers reset when connecting to their serial port. This is something that OctoPrint cannot prevent. But what we can do, is render a one time message when trying to connect to a printer with a confirmation dialog explaining this potential side-effect.

I do not know how many controllers behave like this but I know that the Prusa MK3 does. I will explain this in the text. At leasst people will be better informed and hopefully prevent this problem.

What do you think of this "solution"?

closed 2018-07-29 by gdombiak

Dernjo, 2018-07-29:
This should be acceptable, I am not sure how your app hooks into octoprint so I cannot offer a better solution. Other apps I have tried connect to octoprint and offer limited functionality, like viewing webcam only and/or adjusting temps. Maybe if you have something that detects if s print is running and only loads only some functionality. If your app is causing octoprint to re-connect to the printer, which would indeed cause a printer reset is there a way to just connect to octoprint without causing the connection to the printer happen again?

Sorry for the delay response

gdombiak, 2018-07-29:
The connect button is for asking OctoPrint to connect to the printer. We have a chicken and egg situation, I cannot know if there is a print job running without connecting.

What creates the reset is the printer itself when something connects to the serial port. When you click 'connect', OctoPod asks OctoPrint to connect to the printer (on the serial port). On some controllers (e.g. Arduino) this triggers a reset.

Since part of the problem is that users do not expect the reset on the printer to happen when you click connect, I added a **one time confirmation dialog** that tells them about a potential reboot when connecting. If they click proceed, at least they are now informed. 

Hope this helps give a better user experience.

Dernjo, 2018-07-29:
Fair enough, I see what you mean. Would there be a possible way to hook into the web UI to get the printer status information? Then you could put a limited function parameter and just relay information that is available from the web UI. I just purely speculating here.

gdombiak, 2018-07-29:
That's an interesting idea. What information would you be interested to see? We currently show most of the same info you see on the main page of OctoPrint. We could add more things that you could find useful.

I like your idea of letting you switch to the browser version of OctoPrint. Plugins is something that you can do there and not in OctoPod (e.g. see how leveled is your bed)

Dernjo, 2018-07-29:
Probably basic information would only be required like:
ETA
Current Print Time
Complete Percentage 
Tool/Bed temps
Ability to view webcam
And ability to pause, resume and cancel prints. 

For a future update you could see if you can set up functionality into Ocoprint Enclosure Plugin (https://github.com/vitormhenrique/OctoPrint-Enclosure) and add in the ability to control the lights via on/off button and/or a slider, turn an enclosure fan on/off via button and control speed via slider and show enclosure temp readings. Just a interesting idea a had :D

gdombiak, 2018-07-31:
Those functionalities are available in OctoPod. :)

I created issue #41 for your nice idea. Thanks for your suggestions. I will see also about making use of installed plugins (if possible)

commit 8fe3a762 (2018-07-29): Added one time confirmation dialog when connecting to printer to prevent losing running print jobs.

Closes #35