gdombiak/OctoPod #36: Feature request: PSU Control plugin support.

TheZeroBeast, 2018-07-27:
Hi,

First off, great work on he app, it is vey impressive.

I would love to have the ability to toggle the psu if plugin “PSU Control” is installed.

A setting to add/remove in app even as not everyone would you it.

At present I have to open octoprint webpage to toggle psu then switch to the app.

Any who, keep up the good work.
Rob

gdombiak, 2018-08-05:
Looked at this plugin and we can implement this feature via its [API](https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/API). :)

TheZeroBeast, 2018-08-05:
Hell yes, that would be amazing 👏👌😁

gdombiak, 2018-09-08:
Rob, I'm adding a THANKS file with people that help OctoPod grow. If you give me you full name I can add you. Thanks

TheZeroBeast, 2018-09-08:
No problem mate 👍
Robert Malcolm McKenzie

gdombiak, 2018-09-08:
Tks 👍