gdombiak/OctoPod #55: Multicam

shaelr, 2018-08-21:
The first webcam loads fine. However the 2nd stream never loads. 

Is there maybe a way (similar to the multicam plugin) to put in the address for the stream?

gdombiak, 2018-08-21:
Hi shaelr,

Are you using the MultiCam plugin? I assume that yes. Can you do any of these tests:
1. Copy URL of second webcam that you have in MultiCam and paste it in safari on your iOS device. Does it work?
1. Go to Terminal tab and access OctoPrint from that page. Navigate to Control tab and click on the webcams buttons. Does it work?

My guess is that the phone cannot access the camera stream. If URL is a local IP of your network then make sure you are on WiFi in the iOS device. If URL is a public IP then make sure that your router is correctly configured. Once any of those tests work for you (since they test the same thing) then the 2nd webcam will work fine in the app.

shaelr, 2018-08-21:
The adress works perfectly fine from an outside network */webcam/?action=stream is the default cam. And the secondary is */webcam2/?action=stream.... Both work on an outside network and both links are configured the same in  the multicam plugin. However the app still will not load the second stream

gdombiak, 2018-08-22:
I assume then that connectivity works fine. Here are other ideas: 

1. Is this an MJPG camera?
1. Is OctoPrint able to render the 2nd camera? In the Control tab you can click on the 2nd camera and see what it shows
1. Do the same test from the app. Go to the Terminal tab and access OctoPrint from there. Does the 2nd camera work?

shaelr, 2018-08-22:
Yes to all 3.

However to number 1. The cameras are both configured to stream as yuvy

shaelr, 2018-08-22:
same thing. both cams as mjpg, first one streams fine to octopod. 2nd one doesn't.

gdombiak, 2018-08-22:
Hmm, could you flip the cameras and see what happens? Move the 1st to 2nd and vice versa. 

Are you going through a reverse proxy like ngnix? Or are you connecting directly in the local network?

shaelr, 2018-08-23:
the cameras are the exact same model, however ive flipped them and still the same issue. cam 1 streams fine to octopod and cam 2 doesnt.

Using haproxy, but thats more for http authentication on the server. directly into the local network through port forwarding and a url link.

gdombiak, 2018-08-24:
Just to discard this case, are you using a different authentication (ie. user/password) for the second camera?

shaelr, 2018-08-24:
99% sure they're the same. However i will have to confirm that 100%. Honestly I know you flagged it as help wanted... But i dont really feel like that is the case. Something is wrong with how it's grabbing the second stream Imo.

gdombiak, 2018-09-03:
Closing due to inactivity. We can open a new ticket if this comes up again. While this ticket was opened, we implemented #61 and #62 for easier troubleshooting.

closed 2018-09-03 by gdombiak

gdombiak, 2018-09-03:
> Update, with the new update i get this now.... which is strange
> considering it is the same url for webcam1.... only being "webcam" instead
> of "webcam2"
> [image: img_c7ba4410c13c-1]
> <https://user-images.githubusercontent.com/42502053/45003308-1a4a1380-afaf-11e8-9333-3762904cd01d.jpeg>

That’s good info. Do you have 2 cameras hooked up to the same OctoPrint?
Curious how url of second camera looks like that and not an anbsolute url
(http://....)

I’m away from home now so cannot check this but I this that with this info
I know what to change to support several cameras using relative urls. Btw,
if you are running more than one camera hooked up to the same octoprint
then you might want to consider changing that and have a single camera per
octoprint. CPU and bandwidth might affect octoprint and your print
operations.

Thanks
Gaston

shaelr, 2018-09-03:
yes 2 cameras on the same octoprint. cpu usage isnt much of an issue as the resolution and framerates are low enough that it can manage it. 

I tried changing the url to the full address of the public url (which would follow a similar structure to this "http://www.printer.com:1234/webcam2/?action=stream/" with no change...)

gdombiak, 2018-09-03:
Do you have a screenshot when using an absolute url? I expect error message
to be different.

Gaston

shaelr, 2018-09-03:
interesting... now it appears to work. however if i forgoe the "http://" from the absolute url it no longer works and i get the same error. 

are you grabbing the urls differently?

in multicam both cameras were setup as follows.

"/webcam/?action=stream/"
"/webcam2/?action=stream/"

yet webcam 1 worked fine and webcam 2 gave invalid url.

hopefully there is a fix for this, because the stream url is local to octoprint so to have it going outside to come back in is sorta senseless.

gdombiak, 2018-09-04:
The urls that you see in the MultiCam plugin are sent to OctoPod for
rendering. For relative urls I manually build the absolute url but only in
some cases. From your case I see that there are more cases that I can
handle. For now configure an absolute url for the second camera and you’ll
be fine. I’ll see what I can do in the code for an upcoming release.

Sounds good?
Gaston

shaelr, 2018-09-04:
sounds good.

just glad you finally know what the problem is :)