gdombiak/OctoPod #56: Add Cancel/Reset/Resume buttons.

nerkaid, 2018-08-23:
With that buttons, this app would be the best in iOS for control Octoprint. Thanks for this great job!

gdombiak, 2018-08-23:
Hola nerkaid,

When you are printing, there is an (i) icon next to "Printed" where you see the progress bar. If you click on the (i) you will get a window that lets you pause, resume, restart or cancel the print.

Is that what you were looking for?

Gracias
Gaston (de Argentina) ;)

nerkaid, 2018-08-23:
Exactly that! Sorry I didn't realize it. You can close the ticket. 

Gracias por tu gran trabajo, la aplicación es un pasada!! 

Saludos desde España ;)

closed 2018-08-23 by gdombiak

gdombiak, 2018-09-08:
Estoy creando un archivo con los nombres de las personas que ayudaron crecer a OctoPod. Si me das tu nombre te agrego, sino uso tu alias que veo aca. :)

nerkaid, 2018-09-09:
Hola,

 

Luis F. Fernández, o nerkaid, como prefieras!

Gracias! 😉

 

Un saludo,

Luis F. Fernández

Enviado el: sábado, 8 de septiembre de 2018 19:39
Para: gdombiak/OctoPod <OctoPod@noreply.github.com>
CC: nerkaid <nk@nerkaid.com>; Author <author@noreply.github.com>