gdombiak/OctoPod #57: Implement confirmation for Connect/Disconnect

skohls, 2018-08-23:
My printer (Prusa i3 MK3) resets whenever the USB connection to Octoprint is established or torn down.
This seems to be something about the way the controller board handles USB, just the way it is...
For me it would be great to have an extra confirmation step (perhaps something one can toggle on/off in the app's preferences) before a connect/disconnect really is issued - because an accidential reset during a print is really nasty business.

gdombiak, 2018-08-23:
Hi skohls,

I also have an MK3. Such a nice printer. :)

In version 1.2 we added a confirmation dialog before disconnecting. This dialog appears each time you try to disconnect. In version 1.3 we added a message when you connect that appears only one time telling you that the printer might reboot upon connection.

Do you think this is enough? If you think we should add an optional (settings configurable) confirmation before connect we could add it to the todo list.

skohls, 2018-08-24:
Dear gdombiak,

thanks for the quick reply!

I now realize I didn’t ever use the disconnect button - I was always assuming it would behave the same way as the connect function and I was very cautious not to come near it…
But yes, if it isn't too much hassle, since this forces a reset I'd really like an extra step in between. I use the app often to check on a print remotely and there would be no way to recover should I accidentally reset the printer.

gdombiak, 2018-08-24:
Sure, we can add this setting. By default things will work as they do now but users will be able to have a enable/disable confirmation dialog before connecting and same for disconnecting.

Sounds good?

skohls, 2018-08-24:
Sounds great, thanks a lot!

skohls, 2018-09-09:
> I'm adding a THANKS file with people that help OctoPod grow. If you give me you full name I can add you. Thanks

Hm - that’s way too much credit for me overlooking that half my issue didn’t exist- =^_^=
But if you insist…

Stephan Kohls

Thanks!

Best regards,
Stephan

commit 76b051ee (2018-09-22): User can now control if they want a confirmation dialog on connect/disconnect.

Closes #57
closed 2018-09-22 by gdombiak