gdombiak/OctoPod #64: Add basic auth support

KissmanCZ, 2018-09-03:
Hi again, could you please add basic authentication support?

gdombiak, 2018-09-03:
Hi Jan,

In the printer setup you can specify the user/password to use for authentication. It should work fine with Basic auth and for Digest auth. Is it not working for you?

Thanks,
Gaston

KissmanCZ, 2018-09-03:
I have more user in OctoPi, so I need put different set of credentials for each.

gdombiak, 2018-09-03:
Hi Jan,

Could you confirm your setup to make sure we are talking about the same things? :) Are you configuring the HAProxy that comes with OctoPi to provide HTTP authentication before reaching OctoPrint? Is there only 1 reverse proxy (e.g. HAProxy) before OctoPrint that you are using?

When you say "I have more user in OctoPi", does it mean that HAProxy has been configured to request HTTP authentication and that its userlist has been configured to have many users (https://discourse.octoprint.org/t/how-to-enable-http-authentication-on-haproxy/180)?

If my last assumption is correct, then each person could have its own OctoPod installed. In the printer setup, each person would enter their user/password that will be processed by HAProxy. Once we passed the reverse proxy (e.g HAProxy), we just need the API key to execute commands in OctoPrint. Users/password in OctoPrint are not used since the app uses the API key. Is this your setup and are you running into issues?

Thanks,
Gaston

gdombiak, 2018-09-03:
> So credentials in printer setting are for basic auth instead of OctoPrint
> auth.? Ok, I tried enter my HAproxy credentials into these fields and still
> no luck.
>
> I have HAproxy with user / password and Octoprint with user / password.

Correct, user and password in the printer setup are used for answering any
http authentication request coming from anything in front of octoprint
(e.g. ngnix or haproxy).

Do you see any error message appearing? What does it say? If credentials
are invalid you should see a message saying that.

Thanks
Gaston

KissmanCZ, 2018-09-08:
Yes, solved, maybe pls add some "tooltip" that credentials input in settings is for basic auth. and not for OctoPrint :) some users could be confused (like me).

closed 2018-09-08 by KissmanCZ

gdombiak, 2018-09-08:
Good to hear and good feedback. Will rename the label to "HTTP Authentication".

Thanks
Gaston