gdombiak/OctoPod #65: Allow to set up the feed rate

jollino, 2018-09-03:
Hi there!
OctoPod is definitely one of the best OctoPrint clients out there, but I can't seem to find any way to set the feed rate from it (short of sending an M220 command via the terminal).
Am I missing something, or is it actually not available? Will it be added?
Thanks!

(Slightly OT: if you want, I can translate the app's strings into Italian. I'm a professional freelance translator and I'd be more than happy to give back to the project!)

gdombiak, 2018-09-03:
Ciao Daniele,

Thanks for the kind offer. I will contact you when I get to issue #17 for multi language support. In the panel tab, you can control the flow rate for the extruder. Is that the feed rate you are looking for or is it something else?

BTW, there seems to be no way to read the current flow rate value but to only set it. I had to start with some default value just like OctoPrint does.

Grazie,
Gaston

jollino, 2018-09-03:
Hi Gaston,
I was referring to the actual feed rate, which is the multiplier the printer applies to movements. In OctoPrint, you can find it on the Control pane, at the bottom left:
[image]

It's essentially the same as rotating the controller wheel or using the up/down buttons on the printer when using Marlin and Marlin-derived firmwares, but it can be handy to set remotely; I sometimes need to slow down the first layer with some nasty filaments.

It looks like OctoPrint just sends an M220 Sxxx command, where S is a parameter generally going from 50 to 150 (with 100 as the default): http://marlinfw.org/docs/gcode/M220.html

There seems to be no way to read the current value (in fact, changing it directly on the printer doesn't affect OctoPrint) but most of the time it's just left at 100%, so that would be a safe default in my humble opinion.

Do let me know about the translation!

gdombiak, 2018-09-03:
Excellent info. I had no idea of this other setting. Will file it for
upcoming release.

Thanks
Gaston

closed 2018-09-05 by gdombiak

gdombiak, 2018-09-29:
Ciao Daniele, I just pushed changes so we can add new translations for OctoPod. Let me know if you are still interested in helping add the Italian translation. I added the Spanish (Latin America) as an example.

Grazie,
Gaston

jollino, 2018-09-29:
Absolutely! Since I've never done pull requests and I'd have to read up on that, if you want I can "seed" the translation by just sending you the file, if that's okay.

gdombiak, 2018-09-29:
Sure. Can you send me your email to gdombiak AT gmail.com? Thanks! :D

commit d9f74542 (2018-09-05): Implemented feed rate control.

Closes #65