gdombiak/OctoPod #73: Disconnect cancels print

bernardhny, 2018-09-10:
I started a job for the first time via Octopod. About 37 hours into the job the application video was glitching by showing half video feed and blacked out. So I clicked disconnect thinking that would resolve the problem on reconnect. Instead the print job stopped and the nozzle stayed where it was and remained at temp.

Second issues is with switching printers. When I do , the time remaining glitches showing different times switching every second. The only way to solve the problem is close the app completely and relaunch. This is repeatable.  Any suggestions?

gdombiak, 2018-09-14:
Hi Bernard,

Sorry to hear that happened and sorry for the late reply (been sick for a week). Many printers (not sure which one you have), will reset when OctoPrint connects to the printer. This is why we added a confirmation dialog on disconnect to let the user confirm that he/she actually wants to disconnect OctoPrint from the printer. That seems to not be enough to prevent this confusion as reported by other users so issue #57 was filed to improve this situation. I will try to include it for the next release. Would that have helped with this?

Let me see if I understood the second issue. When you connected to a second printer, the remaining time (and I guess all other fields too) were still showing information about the previous printer and switching to info of the new printer? Like if the connection to the old OctoPrint was still alive and receiving information? Is this something that you can reproduce each time?

Thanks,
Gaston

gdombiak, 2018-10-16:
Hi Bernard,

Today OctoPod 1.11 was released and it includes several improvement regarding connectivity to OctoPrint. Let me know if you still see the problem when switching between printers.

BTW, a few released back, you can configure OctoPod to ask for confirmation to connect to the printer and / or to disconnect from the printer.

Thanks,
Gaston

gdombiak, 2018-11-17:
Hi Bernard,

How are things going with the latest version? I will close this ticket for now. Feel free to open a new one if you are still running into issues.

Thanks,
Gaston

closed 2018-11-17 by gdombiak

umtauscher, 2020-12-16:
Hi Gaston,
thank you very much for your app, I just installed it on iOS and tried it out - sadly during a print in progress.
I am a bit confused, because this issue is so old and still the disconnect stops the print and with that destroys the work in progress.
Just to clear things up "disconnect printer" would in my interpretation disconnect the app from the octoprint instance. Instead it disconnected the printer from Octoprint. Is that correct? I even enabled the read only mode but still it destroys the running print.
I would love to hear from you about this.
TIA
Umtauscher