gdombiak/OctoPod #83: Add support for push notifications

gdombiak, 2018-09-24:
This feature will let users receive notifications (with images if possible) when their print is done. We need to investigate with other events we can notify.

To implement this we will need to create an OctoPrint plugin.

jollino, 2018-11-18:
I noticed an issue: the notifications are always quite a bit late. I also use the Telegram plugin to get real-time notifications every half-hour and upon specific events (start, pause, stop...) and the notifications from OctoPod are always late. Sometimes they're even clustered, e.g. those two that both came at 18:12 but referred to prior prints.
It doesn't affect me much as the Telegram messages are sent from OctoPrint directly, but I imagine that this may be a bit of an issue for those relying on OctoPod alone.

[image]
[image]

Background app refresh for OctoPod is enabled; not sure what else may be causing this!

Hedrakk, 2018-12-18:
For my part, I’m not getting any notification at all. I checked my notification settings and my background app refresh settings and everything seems to be fine. 

Is it because the functionality is still in development? 

Thanks

closed 2019-07-07 by gdombiak