gdombiak/OctoPod #85: Crash when switching printer

iwalldock, 2018-09-25:
App crashes when switching printers in latest update.  Reopening app shows new printer. iPhone X running iOS 12

gdombiak, 2018-09-26:
Hi iwalldock,

Sorry to hear that. I do not see (yet) a crash report from the crash. Crash reports are the easiest and fastest way to understand what happened and fix it. Could you go to 'Settings -> Privacy -> Analytics' and make sure that 'Share With App Developers' is enabled? If enabled already then I will keep checking for Apple to send me the crash report. Once enabled, could you reproduce the crash so a few report gets created and sent?

In the meantime we can try the "old" way. :) Could you give me the list of steps to follow so I can reproduce the problem locally? 

Which version of OctoPrint are you running for each printer?

Thanks,
Gaston

iwalldock, 2018-09-26:
Thanks Gaston!  I’m not sure why you haven’t received crash reports... this setting was turned on already and it has crashed (or closed or whatever it’s doing) at least 30 times since I updated to the latest version of Octopod.  I have three printers, all running Octoprint 1.3.9. All we’re setup and working great with the previous version of your app. When I updated, all you have to do to get the crash is switch printers. Choose ‘Printer’ and then choose a different one. It immediate closes the app and goes to my home screen. When I open the app again, it is now showing the printer that I chose.  I have tried uninstalling the app, restarting and reinstalling. Strangely all of my printer configuration stayed (I’m assuming this is because of your new iCloud functionality). Thanks for the help!

George Hires
iWalldock

gdombiak, 2018-09-27:
Release 1.9 is now available in the app store. Let me know how things go with the new release

Thanks,
Gaston

iwalldock, 2018-09-27:
Works like a charm!  I’m so impressed :)

George Hires
iWalldock

gdombiak, 2018-09-27:
Happy to hear that! Open source rocks! :)