gdombiak/OctoPod #92: Domain Name with '-' Not accepted

hcanchristian, 2018-10-03:
I tried to enter the hostname url which in my case contain a -. This is not accepted by the dialog.

gdombiak, 2018-10-03:
Thanks for reporting this and good catch. Will include a fix for this in the next release. Waiting on 2 more translations and then new release will be out.

hcanchristian, 2018-10-04:
> Antworten an: gdombiak/OctoPod <reply@reply.github.com>
>
>
>
> I'm not being able to reproduce this. Can you give me an example of a URL that is not working with a -?
>
> Tested things like: http://_23.we.com and works fine.
>
> Thanks

Hi Gaston,

 

try http://octo-pi.int.my-domain.de

 

Regards

Christian

gdombiak, 2018-10-04:
Great. With that info I know what to fix now. :)

Testing local fix here and will soon push new version to the app store.

commit d63c187a (2018-10-04): Simplified and fix case that regex for URL was not working. Less accurate to detect valid URLs than before but simpler and still useful.

Closes #92
closed 2018-10-04 by gdombiak