gdombiak/OctoPod #108: Minor UI issues

jollino, 2018-10-14:
Hey there!
I noticed a few very minor UI issues that don't impact the functionality, but are just cosmetic glitches. No rush to fix them!

[image]
[image]

1. My camera has a 16:9 aspect ratio; in an earlier update the ratio was fixed so it's not stretched vertically into 4:3 but there are grey bars filling in the remaining space. I suppose that's the camera UIView; it would be nice if it could be resized to the actual camera aspect ratio / size. It's not much of a big deal on an iPhone 7 like mine, but it would probably be useful on an iPhone SE. One odd side effect on my iPhone 7 is that it makes the lower part unnecessarily scrollable, as shown in the second screenshot.
2. Sometimes the progress bar doesn't seem to be always up to date with the print percentage: in this case it's 0.1% (I had just started the print) and the bar is pretty chunky already. Is there a minimum percentage that's always shown, e.g. it shows a 5% bar until it's actually at 5%?
3. The temperature graph (in the second tab, not shown here) allows for zoom and pan, but I'm not sure how useful that is; sometimes it sort of gets in the way when trying to swipe back and it's quite hard to zoom out to see the full graph again. This is just a personal thing, obviously, but being able to disable pan and zoom may be useful for some users.
4. The File view doesn't give any indication that it's loading when it's initially loading, and with many files on OctoPrint that can take several seconds. Maybe adding a spinner (similar to the one that's shown when swiping down for refersh) would make it clearer that the app is indeed working as intended.
5. I understand that the printer status comes from OctoPrint so it's always in English, though perhaps that could be "translated in code"? E.g. having a localized list of statuses that the string received from OctoPrint can be matched against, and replaced with if one is found.

As always, thanks for the great app!

gdombiak, 2018-10-14:
Love this feedback. Will create an issue for each one as they get done.

Number 4 -> https://github.com/gdombiak/OctoPod/issues/110

gdombiak, 2018-10-19:
Number 1 -> https://github.com/gdombiak/OctoPod/issues/118
Number 3 -> https://github.com/gdombiak/OctoPod/issues/120