gdombiak/OctoPod #113: Print job doesn’t show status

JManDoo, 2018-10-16:
The print job doesn’t show any progress during print

gdombiak, 2018-10-16:
Hi JManDoo,

Thanks for reporting this. Today OctoPod 1.11 was released and it includes several fixes to improve connectivity with OctoPrint when connection went bad. Could you verify that this is still happening with the 1.11 release?

I suspect that this [OctoPrint's issue](https://github.com/foosel/OctoPrint/issues/2850) is what is causing this problem. The fixes in OctoPod 1.11 provide better recovery logic and could improve the situation until [that issue]((https://github.com/foosel/OctoPrint/issues/2850)) gets resolved (assuming that is the root cause for what you are seeing). One way to check this is to go to OctoPrint's logs and search for this:

> octoprint.server.util.sockjs - WARNING - Could not send message to client

If you see those when you saw your problem then high chances for being the root cause. In any case, let me know how the new release that got out today worked for you.

Thanks,
Gaston

JManDoo, 2018-10-16:
Well the progress has never shown for me between two different octopi installs agains two different pi’s and two different iPhones

I always get “Not refreshing” and no temp history either, but I can see the webcam and I can move the x,y and z

Jason

gdombiak, 2018-10-16:
Hey Jason,

That piece of info changes my assumption. :) The “Not refreshing” message could happen because OctoPod is having trouble connecting to your \<hostname you entered\>/sockjs/websocket.

Two possible fixes:

1. Could you check  that your hostname (what you entered when you added a printer) does not end with a /. Example, this is ok "http://octopi.local" and this is NOT ok "http://octopi.local/"

1. Confirm that your reverse proxy (if you have one) includes _proxy_http_version 1.1_ header. Websockets won't work as explained in [OctoPrint's documentation](https://discourse.octoprint.org/t/reverse-proxy-configuration-examples/1107). Previous link explains how to add it and shows examples.

Few more things to do to gather information if the above didn't solve the issue.

1. Go to the Terminal tab and click at the top left on "OctoPrint Web". That will open the regular OctoPrint web interface. Does it work fine? With this we are testing that websockets work fine from the app to OctoPrint

1. Which version of OctoPrint are you running? I assume 1.3.8 or 1.3.9. Just checking it is not a super old version.

1. Could you go to OctoPrint logs and paste here any WARNING or ERROR entries that you see in the log?

1. What do you see if you go to  \<hostname you entered\>/sockjs/websocket in your browser? Could you paste here the text that you got?

Thanks,
Gaston

JManDoo, 2018-10-16:
OctoPrint Version 1.3.9

Going to http://cr10.local/sockjs/websocket returns
Can "Upgrade" only to "WebSocket".

All access is on the same subnet 10.0.0.0/24

Haproxy.cfg

global
        maxconn 4096
        user haproxy
        group haproxy
        log 127.0.0.1 local1 debug

defaults
        log     global
        mode    http
        option  httplog
        option  dontlognull
        retries 3
        option redispatch
        option http-server-close
        option forwardfor
        maxconn 2000
        timeout connect 5s
        timeout client  15min
        timeout server  15min

frontend public
        bind :::80 v4v6
        bind :::443 v4v6 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint

backend octoprint
        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqrep ^([^\ :]*)\ /(.*) \1\ /\2
        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }
        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }
        option forwardfor
        server octoprint1 127.0.0.1:5000
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

backend webcam
        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2
        server webcam1  127.0.0.1:8080
        errorfile 503 /etc/haproxy/errors/503-no-webcam.http

  2.  Confirm that your reverse proxy (if you have one) includes proxy_http_version 1.1 header. Websockets won't work as explained in OctoPrint's documentation<https://discourse.octoprint.org/t/reverse-proxy-configuration-examples/1107>. Previous link explains how to add it and shows examples.

  2.  Which version of OctoPrint are you running? I assume 1.3.8 or 1.3.9. Just checking it is not a super old version.
  3.  Could you go to OctoPrint logs and paste here any WARNING or ERROR entries that you see in the log?
  4.  What do you see if you go to <hostname you entered>/sockjs/websocket in your browser? Could you paste here the text that you got?

gdombiak, 2018-10-16:
Thanks Jason for getting that info. Let's verify if OctoPod can establish a websocket connection to OctoPrint. If you are using OctoPi then you will need to ssh into the box and go to /home/pi/.octoprint/logs. 

1. Close all apps connected to OctoPrint. Also close any browser connected to OctoPrint
1. If you know how to do a tail then use it to monitor the log. If not check content of octoprint.log  (look at the end to see what gets added after this step)
1. Open OctoPod
1. You should see this added **octoprint.server.util.sockjs - INFO - New connection from client: ...**
1. Press power on the phone to turn off the display
1. You should see this added **octoprint.server.util.sockjs - INFO - Client connection closed: ...**

Did you get those entries in the log? Do you see anything else between those 2 new entries?

We can use Facebook messenger if you want to contact me via facebook. Name Gaston and last name Dombiak. Chat will make it faster than this.

gdombiak, 2018-10-16:
Jason figured it out. It was a problem with SSL cert. I will open an issue so that we can expose the exception that OctoPod found trying to connect to websocket. This will make it easier to troubleshoot things

closed 2018-10-16 by gdombiak

Crueltism, 2018-11-21:
Having the same issue. Can somebody explain how to fix this SSL cert. thing ?

kaartz, 2022-02-02:
I am having the same issue. What was the fix for this problem?