gdombiak/OctoPod #117: Add Turkish translation

gdombiak, 2018-10-17:
Add Turkish translation

Alpdo, 2020-01-12:
I saw the original comment, if the commenter didn't respond I can help with the translation too.

gdombiak, 2020-01-12:
Hey @feoranis26 , this is great new and perfect timing. I'm now working on the next major release and adding a few UI changes that require more translation. If you use Facebook could you send me a message? Most translators interact with me via Facebook messages. We can use email as a fallback.

Have you read [How do I add a new translation to OctoPod?](https://github.com/gdombiak/OctoPod/wiki/How-do-I-add-a-new-translation-to-OctoPod%3F)? You will need an XLIFF editor if you do not have one already.

Thanks,
Gaston

jollino, 2020-01-12:
Gaston, I was thinking... while we figure out a way to centralize the translations, maybe we could create a Facebook group for it?

gdombiak, 2020-01-12:
Love this idea Daniele. Could you help set it up? Did I say how much I like the idea? :) Thanks.Grazie.Gracias

jollino, 2020-01-12:
There we go :) Once you join I'll add you as an admin! https://www.facebook.com/groups/octopod/

feoranis26, 2020-01-13:
I'm sorry but i do not use facebook. If you have any other ways we can use
that but we can use email if that's alright with you.

Gaston Dombiak <notifications@github.com>, 12 Oca 2020 Paz, 20:07 tarihinde
şunu yazdı:

feoranis26, 2020-01-13:
Maybe whatsapp?

Alp Doğaner <alpdoganer26@gmail.com>, 13 Oca 2020 Pzt, 15:52 tarihinde şunu
yazdı:

feoranis26, 2020-01-13:
I'm sorry but i don't have facebook. Maybe whatsapp?

gdombiak, 2020-01-13:
Send me an email to gdombiak at gmail and I will set you up.

Thanks again for your help.
Gaston

gdombiak, 2020-04-29:
Closing ticket due to inactivity. Can reopen or create a new one once a translator is found.

closed 2020-04-29 by gdombiak