gdombiak/OctoPod #135: Add Apple TV support

KissmanCZ, 2018-11-10:
Hi, now we have Apple Watch and would be nice to have Apple TV app too. :) what you think?

gdombiak, 2018-11-10:
I think that technically speaking this is a great idea and excuse to learn how to build an Apple TV app. :) No idea how many people would use it to control from the Apple TV versus the phone or the watch. Still I love it!

KissmanCZ, 2018-11-11:
I am thinking about checking status and video stream only :) lot of us starts print at nigh and than go to watch some movies or series. You can check status of print between them. And would be nice to have stream with temperature graphs on Apple TV in classroom in school :) just stream, temp graphs and emergency shutdown button :)

gdombiak, 2018-11-11:
Jan,

This is perfect! Thanks for being clear about what you have in mind and which features would help users of this new app. As development of the phone and watch app starts to slow down, I will look into this cool idea. Maybe I will end up removing the dust from my unused Apple TV. :) 

Samsung Smart TV features ended up killing my need to use Apple TV (I have the old generation).

jollino, 2019-12-27:
Throwing this out there, but how about chromecast support too? :)
https://developers.google.com/cast/docs/developers

ScharV, 2020-02-17:
Hey guys, love this Great idea. I‘am searching for an TV App like octopod.
What‘s the intermediate with AppleTV extension?

gdombiak, 2020-02-17:
Will see if I have time to look into this

ScharV, 2020-02-17:
@gdombiak sounds great 😍

gdombiak, 2020-02-18:
Started to look into this. Learning how to make an AppleTV app. I think that I can make it happen. List of printers will be automatically picked up from iCloud. OctoPod already synchronizes lists of printers between devices so I can reuse that for the AppleTV. No need to configure or add printers! Will see of supporting multiple printers and also multiple cameras and in full screen. Finally, will see if possible/easy to support Siri for requests like, “Hey Siri, show me the 3D Printer status?”.

I started a reddit thread to get more feedback: https://www.reddit.com/r/3Dprinting/comments/f5n3zs/would_you_be_interested_in_an_appletv_app_to/

ScharV, 2020-02-18:
That would be really nice.

closed 2020-03-18 by gdombiak

MacIsUs, 2023-01-07:
Just wondering if there is a way to change the order of the printers? I’d like to move one from 2nd page to first page and the first one in top right of page one I use the least. Thanks!!

commit 4e88ab96 (2020-03-18): Improved layout when there is a single printer

Closes #135