gdombiak/OctoPod #139: Option to "autodial" VPN

KissmanCZ, 2018-11-11:
Hi, would be nice to have "autodial" VPN option in iOS app. What you think? It is pretty anoying to dial VPN manually. Autodetect of external network and if you are out, dial VPN or open VPN setting to dial manually?

gdombiak, 2018-11-11:
Interesting ideas. I assume you are referring to VPN autoconnect. VPN configuration would be done outside of the app but somehow the app would read info from KeyChain and use it to establish a VPN connection automatically. Is this what you had in mind?

Some links that would help implement this:
1. http://blog.moatazthenervous.com/create-a-key-chain-for-apples-vpn/
1. http://blog.moatazthenervous.com/create-a-vpn-connection-with-apple-swift/

KissmanCZ, 2018-11-11:
Yeah... exactly. VPN profile will be set in native iOS settings and app will just connect using this VPN connection.

jollino, 2018-11-19:
I think the issue here may be that the Cisco app also sets up the VPN (OpenVPN does the same, as does the new 1.1.1.1 app by Cloudflare and other VPN apps) so it's also able to connect to it.
I'm not entirely sure but it may be just beyond the reach of any given app to access a VPN set up by another app.
Reference: https://developer.apple.com/documentation/networkextension/nevpnmanager

KissmanCZ, 2018-11-19:
So, we must redefine feature request to the "Add VPN support"? 💃

mvasilakis, 2019-02-25:
I'd love this too. Outside my internal network I forget to connect to my VPN and am forced to go to settings/VPN and activate a profile. It would be great if a printer can be associated with a VPN profile and automatically switch to that VPN. Or even better a switch within Octopod that lets me activate and deactivate a profile.

KissmanCZ, 2021-01-25:
Hi Gaston, could you please review this feature request again? Looks like iOS finally supporting VPN on demand!

https://developer.apple.com/documentation/networkextension/personal_vpn/vpn_on_demand_rules

gdombiak, 2021-01-25:
Thanks @KissmanCZ for sharing that useful info. Quickly read that link and it seems like I would need to add a new setting so that users can configure when they want the VPN to start. By default it would be when not in Wifi only.

GitHubGoody, 2022-01-26:
@gdombiak If you're looking to prioritize enhancement requests, I'm a +1 on this one.😁