gdombiak/OctoPod #141: System Menu

Muerte1983, 2018-11-14:
Would be very nice for me to get access to the system menu controls.  I have toggles for starting and stopping cam there. That's the only reason octoclient is still installed on my iPhone. 😉
Great work so far! 👏

gdombiak, 2018-11-14:
Gracias Muerte! :)

I will work on this tonight and add it to the next release. This is a good one to have besides custom controls.

Muerte1983, 2018-11-15:
Perfect! Thanks for your fast response. I hope it's not too much work to do this.

gdombiak, 2018-11-22:
Half way there :)

[image]

closed 2018-11-23 by gdombiak

Muerte1983, 2018-11-24:
Great, thank you! 👏💪

commit da025ebc (2018-11-23): Added execution of system commands. Some translations are still pending.

Closed #141