gdombiak/OctoPod #147: Implement Siri shortcuts

jollino, 2018-11-22:
Another idea for "Gaston's world domination" of the Apple ecosystem... :)

Siri commands and shortcuts would be nice to give direct orders to the app via voice (I believe this would also work via Apple Watch), e.g. "Hey Siri what's the status of the print?"

Custom shortcuts handlers could be used for more convoluted things, such as "Hey Siri, prepare for printing" => {set tool to 215°, set bed to 60°, home all axis}.

Resources: https://developer.apple.com/sirikit/

gdombiak, 2018-11-27:
Feature implemented and tested locally. Spanish translation updated. Other translations are still pending. Feature being tested by Beta program

closed 2018-11-27 by gdombiak

Budoman, 2020-02-05:
You mention in the SIRI integration wiki page that "warmup..." is a supported shortcut, however, when I go to create my shortcuts on my iPhone, I only see "cool down" as an activity trigger. Is this a bug or was "warm up" left out for some reason?

gdombiak, 2020-02-05:
Hi @Budoman ,

There are Siri donations for setting bed or extruder temperature to a given temp. However, those Siri donations are made only after you set the temperature from the app once. The reason for this is that the target temp is part of the Siri donation. I see that Apple recently added support for prompting for values when you execute a Siri command so I might be able to change this in the future.

If you already set the temperature from the app then we are at the mercy of the Shortcut app. That app shows recently donated shortcuts when you try creating new shortcuts. If you do a search from the main iphone screen (where you search for apps) you can also do searches on donated intentions. You can execute them from there but you cannot create Siri shortcuts from there. :(

Fuchschr11, 2023-02-10:
Only works if printer is connected manually. Will not autoconnect printer. There is no command for connecting. I have enabled autoconnect on octopod server. Any suggestions?