gdombiak/OctoPod #155: Siri Command Responses

heiko7even, 2018-12-15:
If i ask Siri for the remaining time or set the temperature it always responses with $(Time) or $(temperature) but not with real numbers 🙈

gdombiak, 2018-12-16:
Could you post screenshots of the Siri shortcuts that have recorded phrases and also screenshots of what you get when you run them? That information will help troubleshoot the problem.

Thanks,
Gaston

heiko7even, 2018-12-16:
Sure:

https://imgur.com/a/Om5cTYA

gdombiak, 2018-12-16:
Vielen Dank. That helped figure out what is the reason for the error. Will push a new version tonight to the App Store. Apple usually processes submissions in less than 24 hours so tomorrow noon (PST) might have the new version out with the fix.

commit ee628af8 (2018-12-16): Fixed type in German translation

Closes #155
closed 2018-12-16 by gdombiak