gdombiak/OctoPod #157: Preheat using temps in a gcode file

RafeAldridge, 2018-12-19:
I use this plugin https://github.com/marian42/octoprint-preheat 

When you load a gcode file in Octoprint, there is a new "preheat" button next to the Print/Pause/Cancel buttons. Press that and the nozzle/bed heat up to the (first) temperature in the gcode file. 

I'm thinking that in Octopod you could do similar via the files screen; select a file and have "preheat" as an option, as well as "print".

My printer is in a different building that's usually cold, so the remote on/off and remote preheat is really useful for me. The preheat button means I don't have to remember/manually enter temperatures.

Thanks.

Stady234, 2019-12-21:
Any updates? Current version still has no access to the preheat plugin. This would be awesome to get working!!

gdombiak, 2020-01-21:
Hey @Stady234, could you help provide more info [in this issue](https://github.com/marian42/octoprint-preheat/issues/18). The plugin developer needed more context to understand the need. I haven't worked on this any further.

Stady234, 2020-01-21:
I will post context of use on the other ticket thread....