gdombiak/OctoPod #158: TP Link Smartplug doesn’t seem to be working

Hedrakk, 2018-12-19:
Hi,
I can see the TP Link Smartplug plugin, but clicking on it, trying to « slide it », doesn’t seem to do anything. The plugin is working right from the Octoprint web interface, but I would like to be able to do the same in the OctoPod app. 

Thank you.

Hedrakk, 2018-12-20:
> Hi Hedrakk,
>
> Couple of questions to gather more info and help troubleshoot this.
>
> In OctoPrint, could you go to Settings -> TP-Link Smartplug and enable debug logging? Try again switching things on/off from OctoPod. Then go back to OctoPrint -> Settings -> Logging and paste here the content of the file "plugin_tplinksmartplug_debug.log"
> When you try to turn off the switch from the app, do you get a confirmation dialog?
> Are you getting a dialog after you clicked on turn on/off from the app saying something like "ailed to request to turn power on/off"?
> Thanks,
> Gaston

Gaston,

1. The debug logging was already ON. From what I can seen in the log file, everything seems to be fine. 

2. The actual “button” doesn’t seem to be a button at all: I could be clicking on it all day long and it wouldn’t do a thing. I’m not getting in feedback whatsoever. 

3. As I said, I’m not getting any feedback from the app. Here’s a screenshot:

Hum, I tried to send you once with the screenshot and i got an email delivery error message. I’m trying to send the email again without the screenshot. Please tell me how to send you this screenshot correctly. 

I can’t test it right now because I’m in the middle of a print job, but if you have suggestions, I will try them once it’s finished. 

Thanks. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

gdombiak, 2018-12-20:
Bonjour Jean-Francois,

If button is disabled then nothing will happen when you click on it which seems to be your case. Could you paste in here the content of the file _plugin_tplinksmartplug_debug.log_ before and after you open OctoPod and go to the TP-Link Smartplugs page? I'm trying to see any log entries created when OctoPod requested the status of the plugs.

Buttons are disabled when OctoPod failed to get the status of the switch. Since state is unknown, to OctoPod, then we do not know if we should offer the option to turn it on or off. An improvement I could do is to display next to each plug any error message so we can troubleshoot things easier.

Merci,
Gaston (French name but don't speak French)

gdombiak, 2018-12-21:
Thanks. That information helps. This part looks suspicious: 

> [2018-12-20 17:16:16,780] DEBUG: «
[1]Ê"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.5 Build 171206
Rel.085954","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(US)",
"mac":"50:C7:BF:00:54:C2","deviceId":"8006470C5A65A52B396E8795E8CB52F716C1FE
EE","hwId":"60FF6B258734EA6880E186F8C96DDC61","fwId":"0000000000000000000000
0000000000","oemId":"FFF22CFF774A0B89F7624BFC6F50D5DE","alias":"Wanhao
Di3","dev_name":"Wi-Fi Smart Plug With Energy
Monitoring","icon_hash":"","relay_state":1,"on_time":183313,"active_mode":"n
one","feature":"TIM:ENE","updating":0,"rssi":-74,"led_off":0,"latitude":45.5
34626,"longitude":-73.939842}},"emeter":{"get_realtime":{"current":0.213500,
"voltage":120.206845,"power":14.224610,"total":7.583000,"err_code":0}}}

Look at the text after _DEBUG:_  and before _"system"_. If that is the actual text then that is the problem since that is not valid JSON. Could you confirm this is not a copy/paste corruption but the actual text in the log file?

This is an example of what I have in my log

> [2018-12-21 19:27:12,120] DEBUG: {'emeter': {'err_code': True}, 'system': {'get_sysinfo': {'relay_state': 3}}}

I would expect your entry in the log to look like this:

> [2018-12-20 17:16:16,780] DEBUG: {"system":{"get_sysinfo":{"err_code":0,"sw_ver":"1.2.5 Build 171206
Rel.085954","hw_ver":"1.0","type":"IOT.SMARTPLUGSWITCH","model":"HS110(US)",
"mac":"50:C7:BF:00:54:C2","deviceId":"8006470C5A65A52B396E8795E8CB52F716C1FE
EE","hwId":"60FF6B258734EA6880E186F8C96DDC61","fwId":"0000000000000000000000
0000000000","oemId":"FFF22CFF774A0B89F7624BFC6F50D5DE","alias":"Wanhao
Di3","dev_name":"Wi-Fi Smart Plug With Energy
Monitoring","icon_hash":"","relay_state":1,"on_time":183313,"active_mode":"n
one","feature":"TIM:ENE","updating":0,"rssi":-74,"led_off":0,"latitude":45.5
34626,"longitude":-73.939842}},"emeter":{"get_realtime":{"current":0.213500,
"voltage":120.206845,"power":14.224610,"total":7.583000,"err_code":0}}}

If that is the actual text in the log file then I would recommend contacting the [plugin developer and open an issue](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues) to see if that is corrupted JSON? 

Could you check one more thing? Go to _octoprint.log_ and paste here the entries for _octoprint.plugins.tplinksmartplug_. I wonder if there are 500 errors.

Thanks,
Gaston

Hedrakk, 2018-12-22:
I know, the code looks weird... it’s not the same when I look at it from Notepad++ and when I paste it in an email. It would be easier if I could simply attach the log file to this email. Is it possible? When I tried with a screenshot, it didn’t worked. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

gdombiak, 2018-12-22:
Have you tried dropping the file on the comment window as described in here: https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/ ? If that does not work then I can give you my email

Hedrakk, 2018-12-22:
I'm replying from the desktop interface, on my laptop, it should be easier to attach any file. I was replying to you through my mail app... I guess github doesn't like attachments sent using email app.

Anyhow, here's the log file in question and the screen shot of OctoPod TPLink "not so button" button :)

By the way, even thought I'm printing and I'm in read-only, the button is the same whatever I'm printing or not.

I hope it will help.

[image]
[plugin_tplinksmartplug_debug_Before_OctoPod_Launch.log](https://github.com/gdombiak/OctoPod/files/2704842/plugin_tplinksmartplug_debug_Before_OctoPod_Launch.log)
[plugin_tplinksmartplug_debug_After_OctoPod_Launch.log](https://github.com/gdombiak/OctoPod/files/2704843/plugin_tplinksmartplug_debug_After_OctoPod_Launch.log)

gdombiak, 2018-12-23:
Yep, button is disabled so clicking on it does nothing. Could you attach your _octoprint.log_ file? We can see there a little more about what happens in OctoPrint when the TP-Link plugin receives API calls. I think they will appear with this text _octoprint.plugins.tplinksmartplug_. 

I think that I will add a text that displays the reason for being disabled. It will help users self-troubleshoot this problem.

Thanks for your patience.
Gaston

Hedrakk, 2018-12-23:
I agree. I won’t be able to send you the log file right now, I’m kinda busy with Holidays preparations. But I’m making a note to send you the file as soon as I can. 

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

Hedrakk, 2018-12-26:
Here's the octoprint.log file

[octoprint.log](https://github.com/gdombiak/OctoPod/files/2710971/octoprint.log)

gdombiak, 2018-12-26:
The info that we need is not present in the log. When I swipe the panels in OctoPod and I get to the TPLink Smartplugs window, I get these 2 entries in octoprint.log:

> 2018-12-26 15:48:53,671 - tornado.access - INFO - 200 POST /api/plugin/tplinksmartplug (192.168.252.105) 10006.37ms
2018-12-26 15:48:53,672 - tornado.access - INFO - 200 POST /api/plugin/tplinksmartplug (192.168.252.105) 10006.69ms

I'm trying to look into the number before the POST to confirm that it's returning a 200 and not a 500. Let me know once you see the _POST /api/plugin/tplinksmartplug_ entries in the log.

Thanks,
Gaston

Hedrakk, 2018-12-27:
I have a log file here that doesn't contain _POST /api/plugin/tplinksmartplug_ but as several instances of **tornado.access** and all of them have Warning with them. I don't know if it's related, but you can have a look at it...
[octoprint.2018-12-25.log](https://github.com/gdombiak/OctoPod/files/2711081/octoprint.2018-12-25.log)

gdombiak, 2018-12-27:
The 404 are expected since those URLs are invalid in OctoPrint. I guess something on your network is making those requests. The only 409 is also expected since the printer was offline when you collected the log. 

Could you do this final test to see if we can collect the info we need?

1. Close OctoPod
1. Stop OctoPrint
1. Delete octoprint.log
1. Start up OctoPrint
1. Open OctoPod
1. Go to  the TPLink Smartplugs window
1. Go back to the temperature window
1. Go to the TPLink Smartplugs window again
1. Check OctoPrint's log. It should appear the entries we are looking for

Thanks,
Gaston

Hedrakk, 2018-12-27:
Here's the OctoPrint Log file after the described steps. But it doesn't seem to have the lines you are looking for. Have a look.

[octoprint.log](https://github.com/gdombiak/OctoPod/files/2712484/octoprint.log)

gdombiak, 2018-12-28:
One last test (before running out of ideas) :(

1. Make sure that OctoPrint is up and connected to a printer
1. Go to OctoPod, go to the Move tab
1. Click on any of the move axis

You should see things like this in the logs:

> 018-12-27 17:11:22,475 - tornado.access - INFO - 204 POST /api/printer/printhead (192.168.252.105) 6.64ms
2018-12-27 17:11:22,476 - tornado.access - INFO - 200 POST /api/printer/printhead (192.168.252.105) 7.31ms

With this test we can confirm that the app can talk via HTTP to OctoPrint and that _tornado_ logs are logging correctly. If they appear fine then it is a mystery why it is not logging the HTTP request to get the switch statuses. If logs for tornado appear and still our request does not appear then we could try using some networking tool that captures HTTP traffic to see what is going on with the mysterious request. Not sure how comfortable you are with these things. 

Gaston

Hedrakk, 2018-12-28:
I don't know if it help solve the mystery, but I finally manage to be able to use the OctoPod TPLink button.

But it seems like the button is not always active. It became active for the first time when I used the plugin from within the Octoprint web interface. Once I turned on and off the TPLink plug within Octoprint, the button became "available" in Octopod. I was able to turn my TPLink plug on and off from OctoPod. Then I closed/killed the app and reopened it. Once again, the button was not functioning anymore.

I redid the OctoPrint cycle and the OctoPod button "availability" came back again. There is a problem though. I'm getting warning messages, either when turning on and off about failed requests, but the plug do turn on and off. It seems like a false warning message, but maybe there's a reason behind that.

So here's some screen shot of the warning messages and the log files.

[image]
[image]
[image]
[octoprint.2018-12-28.2.log](https://github.com/gdombiak/OctoPod/files/2715085/octoprint.2018-12-28.2.log)
[plugin_tplinksmartplug_debug.2018-12-28.log](https://github.com/gdombiak/OctoPod/files/2715086/plugin_tplinksmartplug_debug.2018-12-28.log)

gdombiak, 2018-12-28:
Thanks for this info.

Let me share this piece of information (ignore if you knew how OctoPrint works). OctoPrint has 2 "_communication channels_". 1) You can make **HTTP requests** to execute actions or request information and 2) You can use **websockets** to read information from OctoPrint. TPLink Smartplug plugin uses **HTTP requests** to toggle switches and uses **websockets** to report status of the switches.

Having said that, did the printer move when you pressed the move button on any of the axis? If it moved then **HTTP requests** communication channel works fine and for some reason OctoPrint's log is not logging **tornado.access** entries since they do not appear in the log as you correctly observed. More about logging network traffic below....

This is what we learned from your last test: when you clicked the TP Link switch button from OctoPrint itself, OctoPrint sent updated switches information via the **websockets** channel. This is why OctoPod shows correctly the status of the button. However, when you pressed the button from OctoPod, OctoPod used the  **HTTP requests** communication channel and this failed as seen by the dialogs that you saw in the app.

Summary of what we know so far: 
1. Your  **websockets** communication channel works fine
1. Not clear if your **HTTP requests** communication channel is working fine. If the test of moving the printer's axes from OctoPod worked then this channel is working fine.

Since OctoPrint is not logging HTTP traffic for some reason, we will need to resort to other ways to capture this information and finally get to the root of this mystery. :)

If you know how to capture network traffic this ignore the rest of this response. There are different tools that you can use to capture network traffic. You will need to ssh into the octopi box for all of them. I do not have a raspberry pi with me now so giving a few options to try.

I think that _TCPDUMP_ is already available on raspbian. Follow [this guide](https://sites.google.com/site/jimmyxu101/testing/use-tcpdump-to-monitor-http-traffic) to see how to log HTTP traffic. Might need to add _sudo_ at the beginning if not enough permission. If this tool is not already installed then [this guide](http://www.ronnutter.com/raspberry-pi-intro-to-tcpdump/) will help.

If that tool is not working for you then you can try using [justniffer](http://justniffer.sourceforge.net/#!/install). If install is giving some errors then [this other guide](http://www.shaddack.twibright.com/projects/hw_3DprinterDevRig/#Installtooltomonitorhttprequests) has more info that might help.

Last tool to try is _httpry_. If this tool is not already available in Raspbian then you can compile it from source:
> sudo apt-get install gcc make git libpcap0.8-dev
git clone https://github.com/jbittel/httpry.git
cd httpry
make
sudo make install 

You then use it like this:

> sudo httpry -i \<network-interface\>

Hedrakk, 2018-12-28:
To answer you question about the move, yes it does when I push displacement buttons in OctoPod.

Unfortunately, I don't think I'll be able to log the HTTP traffic. I've taken a look at the different options you mentioned and It's beyond my level: It could be written in Russian I couldn't say.

Oh well, 98% of Octopod is working awesome for me, I think I will have to live without the support of my SmartPlug.

Thanks again for your help.

gdombiak, 2018-12-29:
Great news Hedrakk!

I got to the bottom of this. The TPLinkSmartplug plugin has been recently updated and the way to interact with it (i.e. its API) has changed and is no longer compatible with OctoPod. I updated OctoPod to use the new API of the plugin and things work fine again.

I also learned how to make OctoPrint log those "missing" tornado entries. No longer needed but sharing info just in case. Go to OctoPrint -> Settings -> Logging. Under Logging Levels you need to add a new row for "tornado.access" using level "INFO". Save the new settings and you are good to go. 

If you want I can let you know once the new beta is ready to be tested so you can confirm that things work fine for you. I will need a way to reach out to you (eg. email, facebook) to share a link with you.

Thanks for bringing this issue up.

closed 2018-12-29 by gdombiak

Hedrakk, 2018-12-30:
Great news indeed! I was ready to give up on this feature. 

Thanks a lot for your determination. 

How can I share my email with you so not everyone will see it?

Jean-Francois Chabot
__________________________________
Capitaine
Equipe Extra-Life des Bionic Marmots|
Bionic Marmots Extra Life Team|
www.extra-life.org/participant/jfchabot

gdombiak, 2018-12-30:
Send me an email. Same username that you see in github at gmail.com

dus10lee, 2019-04-14:
The plugin works for me but each time I turn on or off the plug I get a message that it failed even though it works.  Anyone else getting this pop up message?  Anyway to fix it?

commit c00dc337 (2018-12-29): Updated to use new (backwards incompatible) API of TP Link Smartplug plugin.

Closes #158