gdombiak/OctoPod #165: Video /camera not showing up

mitchcecg, 2019-01-01:
OctoSky and my browser shows the video but not this app....
Any help???!
THANX
MITCH

gdombiak, 2019-01-01:
Hey Mitch,

What is the error message that you see in the camera section? OctoPod should display the camera URL it is trying to reach and the error that you got. You can click on the URL and Safari will open and try to render that URL so you can keep troubleshooting things.

Gaston

mitchcecg, 2019-01-01:
When I click in the URL it just rotates the screen 90 degrees

[image1.PNG]
THANX
MITCH

gdombiak, 2019-01-01:
If you attached an image to your email reply, I don't think that Github will show your attached image. Can you reply from the web interface and include the screenshot?

If you click on anything else but the URL then you are right, the camera will go full screen and rotate. The error message will tell us what is going on and how to fix it. What's the error message that shows?

Gaston

mitchcecg, 2019-01-01:
The error is 
invalid Camera url
the url is 
http://octo.cecg.com:8080/?action=stream <octopi.cecg:8080/?action=stream>

and it works on the internal network  (split dns) 
[image]

mitchcecg, 2019-01-01:
Solved there was a space after the URL that confused this program

closed 2019-01-01 by mitchcecg

gdombiak, 2019-01-01:
I see. Something is not ok in there. Are you using OctoPi? If you do then the URL would normally look like http://octo.cecg.com:8080/webcam/?action=stream. For some reason, yours have hostname _octo.cecg.com:8080_ twice in the URL and it ends with a > character. Those things are creating an invalid URL that OctoPod is complaining about.

In OctoPrint, what do you see under Settings -> Webcam & Timelapse -> Stream URL? The default URL, when using OctoPi, is _/webcam/?action=stream_. If you have something different, I wonder if things work if you use the default value?

Gaston

gdombiak, 2019-01-01:
Oh, I see that you fixed it!! Good catch! 

Hope you can now enjoy OctoPod. :)

Regards,
Gaston

idjmic, 2020-05-16:
I'm having the same issue, except my camera only works on WiFi.  I'm using all the latest versions of OctoPrint, and OctoPod, I even changed the port for the raspberrypi camera and forward it in my router. The app is connecting fine to the printer even the video feed, but as soon as I leave the local network, only the app works and no video feed. Is there a config file somewhere for OctoPod to change the default 8080 port?