gdombiak/OctoPod #168: Add support for iPad PiP

mdpauley, 2019-02-01:
If possible support for Picture-in-Picture would be great. Being able to monitor the video stream while working in other apps would be helpful. It’s possible that [KKPlayerViewController](
https://github.com/131e55/KKPlayerViewController) could work for this.

andrew-codechimp, 2020-06-06:
AVPlayer supports PiP but see @gdombiak you are using MjpegStreamingKit
Before I tinker, did you discount using AVPlayer for any reason?

gdombiak, 2020-06-06:
Hi @andrew-codechimp, not really. I used what I found at the moment. If AVPlayer works as well then that could be a better choice since it is supported by Apple (and maybe more efficient).

If we replace MjpegStreamingKit with AVPlayer, then we need to make sure we are not breaking Apple Watch. I was not able to stream mjpeg directly on the Apple Watch without causing so much lag that was unusable so I ended up asking iOS to give me the first jpeg from the mjpeg. You'll see the hack in MjpegStreamingKit. Another thing that MjpegStreamingKit was extended to do, via callbacks, was to notify when the first image was received (and I think other events as well) so that listeners could react to that. For instance, to detect the size of the image and adjust screen size to that.

andrew-codechimp, 2020-06-06:
Those do sound like some very specific requirements, I know very little about AVPlayer's capabilities but that gives me a checklist to look into thanks.

gdombiak, 2020-06-06:
@andrew-codechimp , IIRC I added all [these hooks](https://github.com/gdombiak/OctoPod/blob/6204d6d29b5c0c1ebb0213a45b55180305e01d32/OctoPod/MJPEG%20Streaming/MjpegStreamingController.swift#L27) from line 27 to 37. If we want to replace this class with AVPlayer then we will need to make sure that somehow that logic is ported.

Thanks so much for your help!!!
Gaston

gdombiak, 2020-10-10:
This issue is duplicated with this one: https://github.com/gdombiak/OctoPod/issues/383

closed 2020-10-10 by gdombiak

DanHernanTheMan, 2022-03-21:
Sorry for reviving an old thread but was this ever brought to the App Store version and, if so, how do I activate it?