gdombiak/OctoPod #171: Camera shows blank/gray

imro2, 2019-02-23:
When I open OctoPod I see an error "Connection Failed - The data couldn't be read because it isn't in the correct format." with and option to Dismiss. When I dismiss it, OctoPod connects to the printer and everything but camera works. I am using raspberry pi with pi cam.

 I am able to open the camera stream url from phone browser. I am able to log into Octoprint from phone browser and watch the camera stream there. So I know it is configured correctly and there are not issues. 

It does not matter whether I am connected locally or accessing it over internet. I do have basic authentication enabled in HaProxy.

Cesar0415, 2019-07-25:
Did you ever find the solution? Having the same exact problem.

gdombiak, 2019-07-25:
Hey guys,

Could you paste here the URL of the camera that you are using? It SHOULD NOT have _127.0.0.1_ as the hostname or the word _localhost_. Those 2 words mean something like _local computer_. The iPhone will try to connect to itself to get the camera and that won't work. 

Thanks,
Gaston

gdombiak, 2020-04-29:
Closing ticket due to inactivity. We can reopen it or start a new one if this is still an issue

closed 2020-04-29 by gdombiak