gdombiak/OctoPod #172: Data not refreshing

edallas, 2019-03-14:
I has a self signed certificate and have verify certificate off in the settings but still can’t get temperature and other data to update.  When I click on the not refreshing banner at the bottom of the app I get a starscream error. 

Any ideas what’s going on?

gdombiak, 2019-03-14:
Hi edallas,

What is the starscream error that you see? Please include any error code and message that you see.

Thanks,
Gaston

edallas, 2019-03-15:
Hi

The error is Starscream.WSerror error 1

edallas, 2019-03-22:
Also, noticed that clicking off the main screen to another page like setting and then back give a second or so of update (temperature will update) and the stop displaying the Not Refreshing banner again.

gdombiak, 2019-03-22:
Hey edallas,

What you see is the app trying to refresh the data and falling into the same error. 

Regarding the error that you posted above I found this interesting link: https://github.com/socketio/socket.io-client-swift/issues/943. That error means that the HTTP Upgrade is failing. That reminded me of this setup needed by OctoPi: https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107. I wonder if that is the reason why websockets are failing do the HTTP upgrade. In particular, look at the Warning section in that page that talks about this.

Let me know how it goes.

ds00, 2019-03-27:
Hey there,
I have a similar problem. Data not refreshing,
“BAD CERTIFICATE OR SSL PROBLEM”

gdombiak, 2019-03-27:
ds00,

Do you also see the same error **Starscream.WSerror error 1** with the “BAD CERTIFICATE OR SSL PROBLEM” error? Are you using the default/out of the box certificate? If you do then using HTTPS (not HTTP) will produce this error. You have 3 options to solve this:

1. Replace the default/out of the box certificate with a real/signed certificate. Safest option if you want a super secure server.
1. Use HTTP instead of HTTPS. Traffic will no longer be encrypted and man-in-the-middle attack would be possible 
1. Keep things as is and change Security setting in OctoPod to NOT validate certificates. Traffic will still be encrypted though man-in-the-middle attack would be possible

Gaston

ds00, 2019-03-27:
No I’m not seeing the same error. Only the error I posted before. 
“BAD CERTIFICATE OR SSLPROBLEM”

gdombiak, 2019-03-27:
Gotcha. Try one of the 3 options I mentioned and you will be good. Choose the best one for you based on your comfort level.

ds00, 2019-03-27:
Hmm ok, can you explain a bit how I can adjust this. Little bit of a noob in the network stuff exualy

gdombiak, 2019-03-27:
In that case, I would go for option 2 or 3 that are very easy. 

For option 2, go to OctoPod -> Settings -> Printer and in the second field (hostname) change https:// to http://.

For option 3, go to OctoPod -> Settings -> Security and disable Certificate Validation.

evanquinnalaska, 2019-05-04:
I am having the same problem with the latest 1.3.11 RC 3. 
Http configured. No ssl, enable / disable certificate validation makes no difference. 
Error starscream error code 1.

jollino, 2019-05-15:
I just upgraded Octoprint to 1.3.11 final and I have this issue as well, albeit intermittently: the data on the main page generally updates, but then randomly stops and I get the same Starscream.WSerror message. The Terminal is also not working anymore at all.

All I did was upgrade to 1.3.11 so something must have changed on the server; I found this in the [change log](https://github.com/foosel/OctoPrint/releases/tag/1.3.11), but I'm not sure if this is related at all: _Deprecate appkey API in favor of new bundled Application Keys Plugin. AFAIK was never used by anyone anyhow._

[image]
[image]

turbine2, 2019-05-16:
Just upgraded my octopi version and I am getting the starscream error 1 as well. I have also checked and my connection setting is http rather than https and I have disabled certificate checking.

jollino, 2019-05-16:
> Just upgraded my octopi version and I am getting the starscream error 1 as well. I have also checked and my connection setting is http rather than https and I have disabled certificate checking.

See #175

gdombiak, 2019-05-19:
Closing. Last reported issue was fixed by #176

closed 2019-05-19 by gdombiak