gdombiak/OctoPod #173: Tplink plug-in error

dus10lee, 2019-04-14:
This plugin gives me an error message when turning on/off the plug.  I get the message “failed to request to turn power off/on”.  It appears to still work, but every time I turn on or off the plug this message pops up.  Also, if the octopi is reset the buttons go inactive in the app until I turn on/off the device in octoprint a few times.  I saw someone report an similar issue a few months ago that was resolved.  Maybe tplink made an update to its api again?

gdombiak, 2019-05-24:
Is this error still happening with the new version 2.2? Wonder if #158 fixed this.

dus10lee, 2019-05-24:
Yes it appears to be working now

gdombiak, 2019-05-24:
Great. Closing ticket. Thanks

closed 2019-05-24 by gdombiak