gdombiak/OctoPod #174: Minor UI change suggestions

jollino, 2019-05-15:
These are just very minor usability things, but I think they may be nice to have in future versions as they may help make the UI even more polished!

1. After scrolling to the other pages of the Panel tab, the only way to go back to the first one (the one with the temperatures) is to either scroll back or scroll beyond the last one. A nicer way would be to simply tap the Panel button in the tab bar itself to "rewind" to the first page, much like Instagram does if you scroll the stories view and tap the home button.

2. In the main view, the keyboard shown when setting a temperature is the numbers one, which makes a lot of sense. However it doesn't have an enter key, and the "Set" button is tiny compared to the giant number buttons. I'm not sure how difficult it would be to implement an enter key in the empty space in the lower left corner of the keyboard; it's an odd position for an enter key, but I think it may be handy.

3. The extrude/retract feature currently has a selector between default and custom speed, which requires an extra tap to enter a custom value. Perhaps the selector could be removed and the field pre-populated with the default speed, maybe even as a placeholder text. If the field is empty, use the default; otherwise, use whatever the user typed.

4. I realize this is something very few users would need, but the terminal interaction is a little clunky. (See #162)

5. Screen rotation seem to mess up the full screen webcam:
 a. Tap the webcam so it goes full screen horizontally
 b. Rotate the phone sideways (so the image is straight, whichever side is fine as it will rotate correctly up to this point)
 c. Rotate the phone vertically: now the webcam is shown in the middle of the screen on a grey background. Tapping it fixes it.
The same happens when the screen rotation is locked in control center:
 a. Click the stream so it goes full screen horizontally
 b. Leave Octopod (home button)
 c. Reopen Octopod: same situation as c above

Unfortunatelly I still haven't heard the chance to look into Swift so I can't attempt changing/fixing these myself, but I thought I'd leave a list just in case you want to consider them for inclusion in a future version. Of course, they're all very low priority.

gdombiak, 2019-05-15:
Hola Daniele. Those are great ideas. Life has been crazy over here so had low time to spend on this project. These ideas are a great way to ramp up again. :)

Grazie,
Gaston

jollino, 2019-05-16:
No worries Gaston, and as always thank you for Octopod in the first place!
It's been hectic for me too, and it looks like it'll stay that way for a while longer, hence no Swift for me at this time. :( However, as always, if there's anything else I can help with— I'm here!

gdombiak, 2019-05-18:
Idea 3 got implemented in #183

gdombiak, 2019-08-18:
Idea 1 got implemented in #216