gdombiak/OctoPod #175: StarScream.wserror-Fehler 1 after updating to Octoprint 1.3.11

Coffey72, 2019-05-15:
Gut this error in Octopod after updating. And can’t see print time etc. 
who can help ? 
[image]

ferdaze, 2019-05-16:
I have had this issue for a couple months now.

gdombiak, 2019-05-16:
Hey guys,

That error seems to indicate that websockets is failing to do the HTTP upgrade. Would you mind checking [OctoPrint's configuration guide](https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107) in particular about using the _ proxy_http_version_ header? This is the most common reason for this problem.

Here is another thread in [OctoPrint's support](https://github.com/foosel/OctoPrint/issues/2941) that covers other reasons for this problem. 

Let me know how it goes.
Gaston

Coffey72, 2019-05-16:
> Hey guys,
> That error seems to indicate that websockets is failing to do the HTTP upgrade. Would you mind checking OctoPrint's configuration guide in particular about using the _ proxy_http_version_ [...]

Thanks for your Help, but i think the Picture confuses a little bit. 
The Cam is working. Anything is working on the OctoPrint Web Page. 

The Problem shows up only in the OctoPod App . Camara is running, i can Control the Printer etc. 
The Problem is that the estimated Time gets not displayed , the % gets not updated etc

jollino, 2019-05-16:
Um, I may have found the culprit.

In OctoPrint's API Key Settings there's a checkbox for "Allow Cross Origin Resource Sharing (CORS) Needs restart"; I enabled that and restarted OctoPrint, and now I get the Terminal to stream properly on OctoPod. I don't have the time to prepare a print now but I think it may be the reason it wasn't working.

Note that in my specific case I'm using http (not https) against the local IPv4 address directly and I haven't changed anything about the network settings, other than setting a static IPv4 somewhere and disabling IPv6 so it's not reachable from the outside.

Oddly enough, however, the checkbox is unchecked after a reboot, yet it seems to work.

Not sure if it's just a huge coincidence, but maybe others can test this?

Coffey72, 2019-05-16:
I checked it now. Looks like that this is the solution. Can’t check it in detail because the printer is off, but Octoprint is running.

Coffey72, 2019-05-16:
> Um, I may have found the culprit.
>
> In OctoPrint's API Key Settings there's a checkbox for "Allow Cross Origin Resource Sharing (CORS) Needs restart"; I enabled that and restarted OctoPrint, and now [...]

This Solved it, could test it now. Thanks !

Mario

closed 2019-05-16 by Coffey72