gdombiak/OctoPod #176: Not Refreshing

Neiljclements, 2019-05-16:


jollino, 2019-05-16:
See #175

Neiljclements, 2019-05-16:
My apologies. Should’ve searched the issues first. It happened straight after the upgrade, so I figured they’d messed with something.
Thanks for the super fast reply.

gdombiak, 2019-05-18:
This is a valid error that was introduced when this ticket was resolved in OctoPrint. https://github.com/foosel/OctoPrint/issues/3024

You can use the workaround mentioned by @jollino until I release a new version with the proper fix. OctoPrint increased the security in the websockets area (see CORS) so now OctoPod is running into an issue. I solved it locally but need to test other things before the new release. Will see if I can push new version by the end of this weekend.

commit 3122c44e (2019-05-18): Include Host header to prevent a CORS problem that denies websocket connections. Closes #176
closed 2019-05-18 by gdombiak