gdombiak/OctoPod #188: Octopod Generates Tornado.access Errors when connecting to octoprint.

evanquinnalaska, 2019-06-04:
Octopod Generates the below error: 
2019-06-04 07:51:25,292 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.254.2
2019-06-04 07:51:25,611 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.254.2) 10.07ms
2019-06-04 07:51:26,080 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.254.2) 9.16ms
2019-06-04 07:51:26,549 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-04 07:51:26,751 - octoprint.server.util.flask - INFO - Passively logging in user _api from ::ffff:192.168.254.2
2019-06-04 07:51:26,805 - octoprint.server.util.sockjs - INFO - User _api logged in on the socket from client ::ffff:192.168.254.2

Pi 3b. Octoprint 1.3.11

gdombiak, 2019-06-05:
Hi @evanquinnalaska, could you provide more information about what you were trying to do in the app when the 409 error happened?

Based on [OctoPrint's API documentation](http://docs.octoprint.org/en/master/api/printer.html#retrieve-the-current-printer-state), the 409 error means that the printer is not operational. Maybe it was printing while you tried to control it? Did OctoPod present any error message indicating that the request failed?

gdombiak, 2019-07-07:
Hi @evanquinnalaska, is this still happening with the latest version of OctoPod? 

Thanks,
Gaston

gdombiak, 2019-07-23:
Closing ticket since I think latest fix should have solved this issue. If this is still happening please open a new ticket. Thanks.

Gaston

closed 2019-07-23 by gdombiak