gdombiak/OctoPod #190: Notification sounds on iPhone

H2SO4JB, 2019-06-13:
Hi Gaston,

To begin with, I want to thank you for a wonderful app.

I have a problem with the new alert feature. I only get silent alerts/notifications on my iPhone, no sound. Even when I use the function "Send test notification".

Of course, I have double-checked my notification settings in my phone, but everything is activated as it should. Notification sounds works for all my other apps. Am I alone with this problem, or is this a known bug?

gdombiak, 2019-06-13:
Hi H2SO4JB,

You are the second person that report this. I think I found the reason for this error so will fix it for the next release. 

PS: I never realized of this problem since I use my phone with no sound/ring ... :)

jollino, 2019-06-13:
Please please please use anything but the default iOS sound :) it's frustrating when apps use that and you have to look to see what's going on. IMVHO there should be a specific sound that makes users immediately aware that it's an OctoPod notification. :)
Maybe this can help? https://medium.com/@PeakselDev/top-17-websites-with-free-sound-effects-and-music-for-your-mobile-app-f2ef49cbf9eb

gdombiak, 2019-06-13:
Useful documentation to fix this: [section 'Adding a Sound to the Notification Content' under LocalNotifications](https://developer.apple.com/library/archive/documentation/NetworkingInternet/Conceptual/RemoteNotificationsPG/SchedulingandHandlingLocalNotifications.html)

H2SO4JB, 2019-06-13:
Perfect, then I can stop trying to solve it myself. Then I wait for the next update.
Please, let me know if I can help you troubleshoot.

gdombiak, 2019-06-13:
Ciao Daniele,

Happy to use any sound you like. Give me a link to a few of them. I will create a new Notifications page under Settings in the app so you can chose a few things like choice of sound, same settings you have in OctoPrint's plugin and for the upcoming MMU alerts too.

closed 2019-06-17 by gdombiak

H2SO4JB, 2019-06-18:
After today's iOS update, I can confirm that notifications now work better. But notifications only work if I have the OctoPod app open in my iPhone. If the app is forced closed, notifications not work anymore. This even if I have the "Background App Refresh" option enabled. No other apps I have require the app to be open for notifications to work. Is this possible to fix?

I suspect this issue is related to the other bug I have reported (#194)...