gdombiak/OctoPod #194: Watch app doesn't work without iPhone app running

H2SO4JB, 2019-06-16:
Hi Gaston,
I can't get the Apple Watch app to work without first opening the iPhone app. I mean, if I force close the iPhone app, then the printer status in the Watch app is no longer updated. Usually, Watch apps work independently, without first having to launch the iPhone app.

Do you know about this? Is this a bug, or is there any limitation that affects this behavior for your app only? If it's a bug, can you fix it?  :-)

gdombiak, 2019-06-18:
Hi,

[OctoPod Plugin](https://github.com/gdombiak/OctoPrint-OctoPod) uses silent push notifications which do not work if the user killed the app. Silent push notifications work fine if the app is in the background or in the foreground. It is possible to fix this but it will require more work (not small so will need some time to get it solved). I will also need to figure out how to make it work when including an image and not using silent notifications.

This issue relates to this previous issue: https://github.com/gdombiak/OctoPrint-OctoPod/issues/1

gdombiak, 2019-06-18:
Misread this thread. :) Saw your link from the other thread to this issue and thought it was about notifications.

This is the reply to this issue. I no longer have an Apple Watch 4 but when I developed this and tested it locally, the Apple Watch app should be able to refresh the information when iOS app is out of reach but a known Wifi 2.4 GHz is at reach. The Watch app will first try to use the iOS app since that is faster and if not available then it will try to make a direct connection using Wifi to fetch the data and also the camera image. For this to work, the Apple Watch needs to be able to connect a Wifi using 2.4 GHz that has connected before.

I did not check that it works with cellular since my watch didn't have that.

H2SO4JB, 2019-06-18:
I belong to "the old school" and usually force close my apps every now and then.  :-)

That's why I discovered that the Watch app doesn't work sometimes. Then I discovered that the iPhone app must be up and running for the Watch app to get its status updates. It doesn't matter if the Watch is connected to WiFi or not. By the way, my Watch is connected to WiFi.

If I start the Watch app with the iPhone app force closed, then the Watch app shows no current status. Only the last known status is shown. If I open the iPhone app while the Watch app is open, immediately the Watch app status is updated. In other words, the Watch app can only get its status updates from an open iPhone app.

I have other apps, for example for my Philips Hue lamps, and it doesn't require the iPhone app to be up and running to work. In other words, it works completely independently.

This is not an big issue, but it would have been nice if you could fix it.  :-)

commit 8ffb3ed5 (2020-01-24): Activate WCSession when app is launched (not when app becomes active since it may never happen)

Closes #194
closed 2020-01-24 by gdombiak