gdombiak/OctoPod #197: Add support for chamber temperature

B0rax, 2019-07-01:
Octoprint recently added support for a chamber temperature. Meaning it will show up as a dditional temperature in the graph with a target that can be set via M141.
It would be great to have this in octopod as well.

gdombiak, 2019-07-01:
Thanks for this idea. Will schedule it for the next release. I was not aware of this new feature. :)

gdombiak, 2019-07-05:
Sneak peak:

[image]
[image]

jollino, 2019-07-05:
Any chance to make OctoPod also report the Pinda probe temperature? It shows as "P:xxx" in the temperature report

closed 2019-07-06 by gdombiak

B0rax, 2019-07-07:
Thank you very much!