gdombiak/OctoPod #198: Alert when there are new plugin versions

gdombiak, 2019-07-01:
Would be nice not having to go to OctoPrint's console to find out that there are new plugin versions ready to be installed. This ticket is to research if this is possible

closed 2019-07-06 by gdombiak

jollino, 2019-07-29:
Just a little UX thing here... I had to look at the code to figure out what Ignore and Dismiss do, as it's not immediately obvious. Since we can't update plugins from the app (which makes sense, it may disrupt prints and so on), wouldn't a single button to acknowledge the message on the app be acceptable? It would essentially be a "preview" of the reminder that the user would get from the web version.
Of course I may have completely misunderstood the whole thing. :D

gdombiak, 2019-07-29:
Both buttons do something different. I was not able to come up with a better word. They did not pass the test of my wife either. :(

Dismiss = Close window and do not alert for a day (I think)
Ignore = Never alert again for this update. If a new plugin update appears then display again alert

Let me know if you have a suggestion for better words.

Grazie,
Gaston

jollino, 2019-07-30:
Maybe "Got it" and "Remind me later"? :-)
Or perhaps "Acknowledge" and "Snooze"…

commit 8e008d8b (2019-07-06): Alert when there are new plugin versions.

Closes #198