gdombiak/OctoPod #199: Update icons to use newest design

gdombiak, 2019-07-05:
It is becoming a frequent request from users to update the icons used by the app. Need to find some other free library/site.

If you are reading this and know about icons that are newer and free to use please leave a comment. Thanks

gdombiak, 2019-07-05:
https://remixicon.com/
https://material.io/tools/icons/

pranelio, 2019-07-05:
https://icons8.com/
https://thenounproject.com/term/open-source/

ZacharyKhan, 2019-07-06:
Any graphic designers willing to help make something?

gdombiak, 2020-01-11:
New look with new icons....

[image]

jollino, 2020-01-11:
Would it be possible to add a button to only home X and Y, and Z separately?
There's a lot of space there and sometimes it can be handy to only do a partial home (e.g. when diagnosing issues; I recently had problems with the bearings and I often rehomed the Y axis only with a `G28 Y` before running an endless series of `G1 Yx Fy`

Default = `G28` [may need a W on Prusa to avoid triggering mesh bed leveling, can't check right now]
XY only = `G28 X Y`
Z only = `G28 Z`

gdombiak, 2020-01-11:
Done. Implemented
[image]

Still thinking about dark icons over dark background ... not sure it's that easy to read

jollino, 2020-01-11:
Great!
White-ish icons would definitely be more legible on a dark background, I think. Can that be changed programmatically depending on whether dark mode is enabled or not?

jollino, 2020-01-11:
I was thinking that maybe the section title could be changed to "Axes" (or "Home axes") rather than "X Y Z" so it's more obvious what it refers to, and it's not like the "i" button that apparently many users didn't even think could contain other settings!

gdombiak, 2020-01-11:
How about "Home"? Working now on adding Dark mode support. :)

closed 2020-01-13 by gdombiak