gdombiak/OctoPod #203: Custom Control page does not refresh when switching printers

sjasonsmith, 2019-07-21:
When on the custom control page in the app, it does not refresh when changing printers. To see the controls for the newly selected printer, it is necessary to switch to another page then back again.

All other pages seem to immediately refresh when the printer is changed.

gdombiak, 2019-07-23:
Thanks for reporting this. Will check it out.

Gaston

gdombiak, 2019-07-25:
Jason,

Could you share more information (e.g. steps) on how to reproduce this issue? I tried the following and worked fine:
1. Selected a Printer
1. Slide right to Temp page and then again to Custom Controls
1. Selected another Printer
    1. OctoPod changed page and went back to main page that shows Print Job information
1.  Slide right to Temp page and then again to Custom Controls
    1.  OctoPod initially displays Custom controls of previous printer but then page refreshes and shows custom controls of the correct printer

sjasonsmith, 2019-07-25:
I just checked in the App Store and I am running version 2.6, which appears to be the latest.

When I switch printers, it does not switch back to the main page. It stays on the Custom Control page. This seems to differ from the behavior you see?

I thought it might be related to some printers having no custom controls, so I added controls to another printer.

I am doing the following:

1. Slide to access Custom Controls page. See correct controls for printer 1.
2. Using printer button in top left corner, switch to printer 2.
3. Video feed switches, but bottom portion of screen is unchanged. Still shows custom controls for printer 1.
4. Do one of the following to refresh and see correct controls.
   - Select the same printer again
   - Slide left and right to force the page to refresh

After playing with this with two and three printers, it seems that the custom control page _does_ refresh, but it shows the custom controls for the _previously selected_ printer, instead of the _newly selected_ printer. If I cycle through printers, I always see the controls for a different printer.

gdombiak, 2019-07-28:
Got to the bottom of it. I had different plugins installed in both printers so I was not able to reproduce your case. It is now fixed!

Thanks for bringing this up.

Gaston

commit f2db19f9 (2019-07-28): Connect to newly selected printer before updating subviews. Subviews will now operate on new selected printer

Closes #203
closed 2019-07-28 by gdombiak