gdombiak/OctoPod #207: Add menu button to execute custom control

gdombiak, 2019-07-29:
Allow user to slide to the left to expose _Run_ button when browsing custom controls. If command has no parameters then command will be executed. If command has parameters then windows to complete parameters will be shown (as it works today).

With this improvement we can make it easier to run commands that do not have parameters

jollino, 2019-07-29:
Would the slide be to prevent accidental triggers? I'm thinking that it may be a little non-obvious unless one is aware that it can be slid to the left... maybe an accessory button so it's a small target, harder to tap by accident?

gdombiak, 2019-07-29:
That's an interesting idea. I think I like the idea of having a button so it is explicit that you can execute the command by clicking on the button.

closed 2019-10-05 by gdombiak