gdombiak/OctoPod #211: Webcam URL

dbird86, 2019-08-02:
Is it possible to add a custom URL for the webcam. The app assumes the webcam address lives under the address for the octoprint instance. 

I run 3 seperate instances of Octoprint on one Raspberry Pi 3 so I have seperate addresses for each instance eg: http://<RPi-ipaddress>/printer1/, http://<RPi-ipaddress>/printer2/ and http://<RPi-ipaddress>/printer3/. The webcams live at http://<RPi-ipaddress>/webcam/, http://<RPi-ipaddress>/webcam2/ and http://<RPi-ipaddress>/webcam3/

Octopod automatically assumes the webcam address is: http://<RPi-ipaddress>/printer1/webcam/, http://<RPi-ipaddress>/printer2/webcam2/, etc so fails to display the webcam.

Is this possible to fix?

gdombiak, 2019-08-02:
Hi,

Are you using the [Multicam plugin](https://plugins.octoprint.org/plugins/multicam/) as explained in the [FAQ page](https://github.com/gdombiak/OctoPod/wiki/FAQ-using-the-app#how-do-i-use-octopod-with-multiple-cameras)?

I have your same setup and works fine. Make sure that the multicam plugin is correctly configured and OctoPod will automatically pick up the configuration from OctoPrint.

dbird86, 2019-08-07:
Oh yeah I got it to work, I just put in the full address in the standard webcam settings instead of just the relative path. Now it works in OctoPod :)

gdombiak, 2019-08-07:
excellent. Closing ticket then

closed 2019-08-07 by gdombiak