gdombiak/OctoPod #213: Change extrude by 100 to 50

gdombiak, 2019-08-11:
It is more convenient to control by units of 50 rather than a big 100.

closed 2019-08-11 by gdombiak

jollino, 2019-08-15:
I'm using the latest beta but I still have 100, and it does send a G1 E100. Maybe a localization issue?

[image]
[image]

(Also, is there any way to see the current app version from within the app? I thought it was possible, but I can't find it)

gdombiak, 2019-08-15:
Great detective work. You are right that the translation was still showing the old value. Fixed that and also added OctoPod version to the Settings page. New build is being processed.

Grazie!!!!

jollino, 2019-08-15:
I was wondering... Maybe numbers should be "exempted" from being localizable strings, to prevent this sort of inconsistency in the future? After all a number is a number and iOS should possibly take care of formatting using the current locale (though that's likely not needed at all for an app like Octopod).

gdombiak, 2019-08-15:
Yep, I think that if I remove the "translation" then xcode will show a warning and default to the one in english. The number that we see is actually a string/label so it shows as something that needs to be translated

commit baf7e2a1 (2019-08-11): Extrude by 50 instead of 100

Closes #213