gdombiak/OctoPod #219: Add cell with date and time for expected print finish

simonbogh, 2019-08-27:
Besides the two cells with `Print Time` and `Print Time Left`, for long prints or end-to-end printing, it would be nice to have an additional cell with e.g. `Print Finish: Aug. 28. 10.45 AM` simply by taking `Print Time Left`and adding it to current local time.

gdombiak, 2020-01-13:
Was thinking that since many people use iPhone SE that has a small screen that instead of adding a new row for this information, how about doing a click on the "Print Time Left" row will flip and show "Print Finish:". It will work as a toggle between one and the other.

gdombiak, 2020-01-15:
Implemented like this
[image]

commit d93e4ab0 (2020-01-15): Added row that displays estimate time to completion

Closes #219
closed 2020-01-15 by gdombiak