gdombiak/OctoPod #221: File name on 2 lines in the "Files" tab

slaading, 2019-08-29:
My slicer generates automatic (long!) names with important informations regarding the settings so it could be useful to be able to distinguish 2 different files with slightly different settings.

The 2 lines solution is maybe not ideal for who has hundreds of files (not my case) but I can't think of a better idea.

jollino, 2019-08-29:
Maybe the file name field could be made to scroll, like the infamous `<marquee>` tag of yesteryear's html?

gdombiak, 2019-08-29:
Nice ideas guys! I will see what I can do about this. Tagging it for next release

gdombiak, 2019-08-29:
Sébastien, I think you also mentioned this issue in the "job details" popup in the main page. When you click on the (i) next to Printed. Will see of improving that page too.

slaading, 2019-08-31:
> Sébastien, I think you also mentioned this issue in the "job details" popup in the main page. When you click on the (i) next to Printed. Will see of improving that page too.
> 
> Same for file detail page.

Yes, great !

slaading, 2019-08-31:
@jollino Scrolling is a great idea to save up some space but maybe really hard on the eyes when you have a full page of scrolling texts :)

@gdombiak I could satisfy myself by having the full name only in the file detail page. When in doubt about a file I would only click on it. Easy no?

jollino, 2019-08-31:
> @jollino Scrolling is a great idea to save up some space but maybe really hard on the eyes when you have a full page of scrolling texts :)

Very true, I hadn't thought of that! Maybe the scrolling could be started and stopped on touch?

gdombiak, 2019-08-31:
This is how it's looking so far ... 

Job details has up to 3 lines for file name: (_text translation is still in progress_)
[image]

File details has unlimited number of lines for file name:
[image]

No changes here
[image]

gdombiak, 2019-08-31:
I think that this solves this ticket. Let me know and I can commit the code and close the ticket!

Thanks guys,
Gaston

slaading, 2019-09-01:
Looks great, thank you!
Maybe the « fichier » title could be vertically aligned to the top? ... sorry I’m a designer ;)

closed 2019-09-01 by gdombiak

gdombiak, 2019-09-01:
@slaading that is great feedback. Having a designer on board is awesome! I updated the screen as you suggested.

commit 404976de (2019-09-01): Align File to the top

Closes #221