gdombiak/OctoPod #225: OctoPod noip

pframon, 2019-09-17:
Would be possible to use no-ip address to connect my printer out of home?

gdombiak, 2019-09-17:
Sure. When you enter the printer hostname you can enter an IP address or a hostname. If you enter an IP address and you are out of your home then make sure that you have a static IP address. The default for homes is to NOT have a static IP address given by your ISP. If you go the hostname route then you can use something like no-ip or any other dynamic DNS service out there. Make sure that your router or some computer at home can publish your public IP address to the dynamic DNS service. OctoPod will resolve the hostname and reach your home when you are out of home.

Here is a setup guide from no-ip: https://www.noip.com/support/knowledgebase/getting-started-with-no-ip-com/

pframon, 2019-09-17:
Hi.

Thanks for your quick replay.

I have a dynamic DNS service, myname.ddns.net. I understand that in SETTINGS, PRINTERS, I have to enter my DNS in this form: http://myname.ddns.net/192.168.1.xxx, according your indications. But, when I try to connect, message HTTP:404, and it’s impossible to reach my printer. Even, I try adding a router port, 443 or 80.

What am I doing bad? What is my mistake?

Thanks for your kind helpl.

Regards.

Placido 

Enviado el: martes, 17 de septiembre de 2019 19:53
Para: gdombiak/OctoPod <OctoPod@noreply.github.com>
CC: pframon <pframon@gmail.com>; Author <author@noreply.github.com>

gdombiak, 2019-09-20:
Hola Placido,

La direccion http://myname.ddns.net/192.168.1.xxx no es correcta. Si pones http://myname.ddns.net que es lo que pasa? Para que eso funcione tenes que tener que tu router or alguna de tus computadores este actualizando myname.ddns.net con el IP publico que te asigno tu ISP. Ademas esta suponiendo que el port 80 en tu router esta haciendo forward a tu computadora que corre OctoPrint. Sin esas 2 cosas no va a funcionar.

Saludos,
Gaston

Stady234, 2019-12-21:
Did you ever get this working? Do you still need help?