gdombiak/OctoPod #227: Custom System Commands Execute Error

qkevinto, 2019-10-02:
Having a bit of trouble getting some custom system commands to execute.

I’ve added some custom system commands that does a curl to a webhook to turn some lights on and off, and it works fine when executing via the web interface but doesn’t seem to want to work through OctoPod.

I get a dialog saying “Warning: Failed to request to execute command”:

[image]

Any idea what’s going on here? The standard commands (shutdown, restart, etc.) works fine.

gdombiak, 2019-10-02:
Hi Kevin,

It seems like you added those controls as system commands rather than [custom controls](http://docs.octoprint.org/en/master/features/custom_controls.html). You can add custom controls using this [convenient plugin](https://plugins.octoprint.org/plugins/customControl/). You should be good after that. 

Let me know how it goes.
Gaston

gdombiak, 2019-10-02:
If you want to use them as system commands then check OctoPrint's logs to know the error code that is returning. I see in OctoPrint's documentation that any of these errors may be happening: http://docs.octoprint.org/en/master/api/system.html#execute-a-registered-system-command. Curious to hear which one you are running into.

Gaston

qkevinto, 2019-10-02:
Thanks for the quick response @gdombiak :)

Adding them as a system command was intentional as I wanted to execute a shell command as opposed to sending GCODE to the printer.

BUT! I did manage to figure out the issue. When setting up a system action you set at least three fields, `action`, `command` and `name`. My problem was that I had a space in my `action` identifier:
```
system:
  actions:
  - action: Lights on
    command: curl -X POST https://maker.ifttt.com/trigger/prusa_lights_on/with/key/xxxx
    name: Lights on
```

However, this fixed it:
```
system:
  actions:
  - action: lightson
    command: curl -X POST https://maker.ifttt.com/trigger/prusa_lights_on/with/key/xxxx
    name: Lights on
```

Note that `Lights on` was changed to `lightson`.

gdombiak, 2019-10-02:
Just in case, make sure that the values of action and source do not include white spaces. That may break executing them.

closed 2019-10-02 by qkevinto

gdombiak, 2019-10-02:
Interesting. I was wondering if that was your case. Good to hear you figured it out.

Thanks,
Gaston