gdombiak/OctoPod #229: Notifications: Unknot code

Raekro, 2019-10-18:
Since yesterday, I no longer receive notifications as before. 
Now, whatever the status of the printer, I receive one and only notification: unknow code.
Octoprint 1.3.11
Octopod plugin 0.2.2.2
Octopod iOS v2.8 build 2
I tried to reinstall the iOS application and plugin, but still the same problem.

gdombiak, 2019-10-19:
Hi @Raekro, are you using French language in the application? If yes then I know what is going on and how to fix it. If that is your case, then you will need to change the language in OctoPod to English or anything but French until I get this fixed.

Gaston

gdombiak, 2019-10-19:
@Raekro, I just released Octopod plugin 0.2.3 that includes French translation. Let me know if that was the issue you were running into and if it got fixed for you after updating the plugin

Raekro, 2019-10-19:
@gdombiak Indeed, my application was reviewed in French. In English, everything works.
 
I then tried the proposed new version, then I put the application in French.  For now everything is ok, this version fixes the UnknowCode error well:)

closed 2019-10-19 by gdombiak