gdombiak/OctoPod #230: Problems with video feed on iPhone and Apple watch

FastBlinker, 2019-11-01:
Hello, I’m having problems with my video feed on the Octoprint iPhone and Apple Watch, I get a 404 error for the video feed but temps read fine and progress of print, if I access octoprint from a browser the video stream works fine. Also if I click the info icon it says certificate not valid or ssl problems. Any help would be appreciated.

gdombiak, 2019-11-03:
Hi @FastBlinker,

Are you still running into this issue? Is yes could you please 
1. Describe your setup. For instance:
    1. are you using OctoPi? 
    1. which version of OctoPi and version of OctoPrint
    1. Is this happening when you are in your internal network (e.g. home)
    1. Does it work from internal network but not external network (e.g. outside of your home).
1. share a screenshot of OctoPod when video feed is not working
1. share a screenshot of OctoPod when video feed is not working
1. which icon are you clicking on? Can you also share a screenshot of the certificate/ssl problem

Thanks,
Gaston

FastBlinker, 2019-11-04:
I’m still having the problem, I think it has something to do with the certificate on the Octopi I would just rather not install everything again because I don’t understand much of the internal workings. Not working even over on same network on iPhone the strange thing is that it does read the temperatures just no video. On Apple Watch or iPhone or for that matter with remote access with the actual octopi interface.
 

Sent with the intention of your retention of this extension of hypertext protocol.

gdombiak, 2019-11-04:
Thanks @FastBlinker for the information. IIRC, OctoPi comes with a self-signed certificate so let's try 2 things and let me know how it goes.

1. In OctoPod, go to Settings -> Security and disable Certificate Validation. Self-signed certificates are not valid certificates so we can instruct OctoPod to ignore verifying them.
1. If previous setting didn't fix it then: In OctoPod go to Settings -> Printers and for your printer could you change the URL to HTTP instead of HTTPS? If you are using things inside of your home then you relying on HTTP is enough (IMO).

Those 2 recommendations are based on the certificate issue that you mentioned. If you are hitting the 404 error then I will need more info to help with that.

Thanks,
Gaston

FastBlinker, 2019-11-04:
I’ll give it a try when I’m home tonight, I’d did at one point turn off the verification of certificate in the octopod app but it still wasn’t working, I would need to get the certificate up and running eventually because the whole point was to keep the printer under control remotely 

Sent with the intention of your retention of this extension of hypertext protocol.

FastBlinker, 2019-11-04:
If I turn off validation it makes no difference. 
If I take off the S it immediately tells me unable to connect as soon as I open the program. With https octopod gives me the temp of bed and nozzle and also if it’s printing or not just no video. I can even control the axis’s 

Sent with the intention of your retention of this extension of hypertext protocol.

FastBlinker, 2019-11-04:
With the octoprint interface I see video locally

Sent with the intention of your retention of this extension of hypertext protocol.