gdombiak/OctoPod #231: Palette 2 Plugin 2.3.1 ping valuss

NelsonMB, 2019-11-27:
After updating the Palette 2 plugin to 2.3.1, ping values are not appearing in OctoPod. Please see the attached screenshots. 
[image]
[image]

gdombiak, 2019-11-28:
Thanks @NelsonMB for reporting this issue. I just looked at the Palette 2 plugin and I see that its API got modified. I will update OctoPod app to be able to work with the old version of the plugin and the newest version of the plugin too. I guess that some people still run the older version so need to handle any version.

Thanks,
Gaston

gdombiak, 2019-12-01:
Hey @NelsonMB, I have a new beta version of OctoPod that addresses this issue. If you would like to help test the beta release to confirm it fixes this issue, you can send me an email to gdombiak at gmail dot com.

Thanks

closed 2019-12-05 by gdombiak