gdombiak/OctoPod #232: Notification Settings

Neiljclements, 2019-11-30:
It would be nice if we could choose which notifications we receive please.

Whilst a lot of them are useful, there are a few I don’t need to see like ‘Bed Below Temp Threshold’ 

Is this something that can be done?

gdombiak, 2019-11-30:
Hi @Neiljclements,

Have you tried setting the value to 0 minutes in OctoPrint -> Settings -> OctoPod Notifications -> Bed Notifications? Except for "print done" notifications, all other notifications should be disabled when a value of 0 is set.

Let me know if this is not working as expected.
Thanks,
Gaston

Neiljclements, 2019-12-03:
Watcha @gdombiak ! 
Thing is, I don’t want to kill notifications I’ve  or seen before that are really useful.
I got one today that OctoPrint hadn’t shown before that helped me diagnose an end script problem with the slicer.
I’ll have a play and see what happens.
Thanks for the reply.
I’ll close this now as I’m happy to crack on.

closed 2019-12-03 by Neiljclements