gdombiak/OctoPod #233: iPad & iOS 13.2.3 Seems to induce weird Portrait/Landscape behaviour on Video Stream

ericd1949, 2019-12-02:
This a great App, I've used it for a few weeks now on an iPad4 with iOS 10.3.3 connected to my Octoprint vers. 1.3.12, I carry it around the house whilst printing and can instantly change stuff.

Brilliant!!

Black Friday sales came and a I got an iPad 7th gen, iOS 13.2.3, and on this iPad the Video feed window positioning is weird. Both of my iPads have the same Octopod settings and the iPad portrait/landscape lock is off for both of them.

Here's a sequence of events on the newer iPad that illustrates the "interesting" behaviour:
**Turn on app in landscape
    Panel with no video feed window, top line is the grey bar with "PRINT JOB"
** Now turn iPad to portrait
    Panel with video feed window appears at the top, about 7cm across
    (smaller than the equivalent on the iPad 4 who's video is the full screen's width)
** Tap video window
    Full screen landscape video, iPad is still in portrait position.
** Turn iPad to landscape position
    Now still have landscape video, iPad in landscape position
** Turn iPad to portrait position
    Full screen, side to side, portrait video
** Tap video window
    Small video window & panel - display is portrait, iPad is portrait

There is no sequence that will get a landscape panel with a video feed window.

Full disclosure: up until now I haven't lived in iPad Land very much, I've spent most of my tablet existence in the Wild West that is Android Land.

Thanks very much for your App,
Cheers,
Eric

gdombiak, 2019-12-03:
Hi Eric,

Whenever I have time, I'll try to use the simulator to see what you are seeing with your iPad. Would it be possible for you to post an image of what you are seeing? That might save me some time.

Thanks,
Gaston

ericd1949, 2019-12-03:
Will do Gaston, tomorrow, screenshots of the vital moments will head your way, I'm sure iOS will make it easy for me :-)

Cheers,
Eric

WahWay-ing on my sofa - somewhere ....... 😉

gdombiak, 2019-12-03:
haha. iOS (or iPad OS now) makes it easy. https://support.apple.com/sl-si/HT200289

ericd1949, 2019-12-04:
Wow, that was fun, iPad mail client, where are my files ?

So, with a bit of luck I’ll attach the screenshot of the landscape screenshot with no video window, and also the portrait screen with a very small video window.

Hope this helps,

Cheers,
Eric

jollino, 2019-12-04:
@ericd1949 , you can attach files on iPad's mail client by doing this: https://support.apple.com/guide/ipad/add-attachments-ipadcf744d81/ipados (or perhaps just use the github web page? not sure as I have no iPad)

ericd1949, 2019-12-04:
Thanks for your help, just for now, I've emailed them to myself and pasting them in. Such fun :-)

So, that seems to have worked; first the landscape screenshot with no video window, and then the portrait screen with a very small video window (iPad4 working beautifully in shot).

[image]

[image]

ericd1949, 2019-12-04:
Oh dear I didn't mean to close this, sorry so clumsy of me.

ericd1949, 2019-12-15:
Does my OctoPod (now v5.9) on my iPad 7th gen, iOS 13.2.3 think it has the display characteristics of an iPhone?

I persuaded my friend to load OctoPod on her iPhone 6 running 12.4.3

And the behaviour it exhibited was precisely the same as I detailed above; exactly as you would expect given it's more confined screen. In other words, my iPad7 is behaving the same as an iPhone with regards to portrait/landscape orientation changes, etc.

Oh dear, but I can't resist speculating in an area I have no knowledge of, but, has whatever the signalling protocol the OS uses to talk to Apps vis-a-vis iPad/iPhone identification, changed in 13.2.3?

Cheers,
Eric

gdombiak, 2019-12-15:
Hi Eric,

Two words: Good guess!

OctoPod offers the same user experience on both iPhone and iPad. Depending on screen size of the iphone/ipad, it will decide how big should the image of the webcam be. For unknown screen sizes it defaults to some small value.

I wonder if that is your case. Could you tell me which screen size your iPad has? I think it is a 9.7 inches with a resolution of 1536x2048. I can check wether OctoPod knows how to render it or not.

Thanks,
Gaston

gdombiak, 2019-12-15:
Might be better to know the iPad model. I need to know the screen height. I expect it to be 1024 for a 9.7 inches screen. Maybe this is the problem why you see such a small image

jollino, 2019-12-15:
I cannot say anything about the size of the feed shown on an iPad as I cannot test it, however it may be somewhat related to #118: maybe the video 'thing' has a maximum size that works for the iPhone 11/XR (which I believe is the biggest phone in terms of pixels), but then falls short on the iPad. It's been literally years since I did any iOS development so I'm not really sure how it all works nowadays. I in my days we used Objective-C, not Swift, and even autolayout was a new thing. :)

Regarding the original issue, I noticed a similar oddity on iPhone too:

1. Phone in portrait mode, video above as expected
2. Rotate to landscape, video disappears (as expected, I suppose, though I wouldn't mind a split view if at all possible...)
3. Back to portrait, tap the video feed, full screen video in landscape (with two grey bars on the sides in my case, as the screen is longer than 16:9 on the 11; same as #118 probably)
4. Rotate to landscape, video remains the same or rotates 180° depending on the side
5. Rotate to portait, video is straight but tiny, with giant grey bars above and below; even odder, the status bar is a different shade of grey than both the ones above and below the video, and even differ from my color scheme settings!

I'm not entirely sure what we should expect in this case, maybe just prevent the full-screen video feed from rotating to portrait mode?

https://www.youtube.com/watch?v=hYcgCu3AKvA

gdombiak, 2019-12-15:
Thanks @jollino for the video. That helps. I'm not sure what is the suggestion here for improvement though. Is it about the shade of grey of point 5?

On the iPad, when you do point 2 you still see the video on top since there is space for it. On smaller screens I just hide the video.

ericd1949, 2019-12-15:
Sorry for delay, nipped out for a beer.

This is a, so called by Apple, iPad (7th generation)
Model Number MW772B/A

and according to them it has a:
Retina display
10.2-inch (diagonal) LED-backlit Multi-Touch display with IPS technology
2160x1620-pixel resolution at 264 pixels per inch (ppi)

Wasn't the 9.7 inch" last" years model, and this is "this years, 2019" model.

Let me know if I can help.
Cheers,
Eric

jollino, 2019-12-15:
> Thanks @jollino for the video. That helps. I'm not sure what is the suggestion here for improvement though. Is it about the shade of grey of point 5?

Um if anything I'd probably prevent the full-screen video from showing up in portrait mode, since it's effectively the same size as the main view, plus you can only get there by tapping into landscape and then rotating, so it feels a little weird from a user's standpoint.
I would do this:
a) When in full-screen video mode, allow the video to rotate to the other horizontal direction (like it already does) but disable portrait mode
b) When in normal mode, go to full-screen video mode when the phone is rotated (in addition to tapping). Showing normal mode in landscape on a phone doesn't really make much of a difference, except maybe for the temperature graph.

Just my humble two cents of course!

gdombiak, 2019-12-15:
@ericd1949,

I was able to reproduce. Will fix and submit a new Beta version tonight (if possible). Are you in the beta testers group? If not send me an email to gdombiak at gmail if you would like to be part of it.

Daniele, will get back to you once done with Eric's issue.

Thanks,
Gaston

gdombiak, 2019-12-15:
Fixed. This is 16:9 aspect ratio example
[image]

And here 4:3
[image]

Will now test the other new ipad models of 2019. Had no idea they changed resolutions. :)

jollino, 2019-12-15:
No rush Gaston! I just mentioned it here with no intention of intruding — I hadn’t noticed it until I tried to reproduce Eric’s steps, as I seldom use the full screen view myself :)

commit 73a16df4 (2019-12-15): Fixed camera dimension for iPad (7th generation) (2019).

Closes #233
closed 2019-12-15 by gdombiak